logisim-evolution/logisim-evolution #137: MOSFET and analog component lib

hydrogenbomb1212, 2017-11-03:
logism works fine for simple circuits but as the circuit grows i had to use inefficient subcircuits that could have been way better had i been able to use SRAM made from a mosfet circuit.

i don't have a lot of java experience but ill try to get a optional analog lib going with the future goal to add power draw and analog I/O

AerodynamicBrick, 2017-11-03:
Wow, that's a pretty nifty idea.

maehne, 2017-11-06:
Why don't you simply use the already available RAM and ROM models from the memory library? They are quite configurable. The only limitation is that they currently cannot be synthesised on an FPGA, as no VHDL/Verilog code generator has been implemented for these components yet. Adding this functionality is much less complex than trying to extend Logisim to become a mixed-signal simulator. Pull requests are welcome!

hydrogenbomb1212, 2017-11-06:
well its simple, Sram and Dram have big differences. Sram is more power efficient, lower latency capeble of way higher write speeds and name it, its just more expensive to intergrate. thats the doffirence in a nutshell. sram is better but way more expensive

maehne, 2017-11-07:
I am well aware of the differences of SRAM and DRAM. However, I still don't see why you would like to model that at the transistor-level in Logisim? Logisim as a digital circuit simulator uses the discrete-event model of computation for executing the system model constituted by your schematics. Simulator time advances in a discrete fashion with the clock ticks. Non-zero gate delays are not considered by the simulator. Signals and signal busses only carry logic states (U, 0, 1, X). Under these constraints, only modelling of the latency of your memory may make sense. This you can do today by making a macro-model of your DRAM, where you may model the latency of the outputs with a delay chain of D-FFs.

Also, when you make the step to use a real-world digital simulator using a HDL, such as Verilog or VHDL, you would not go much further in your modelling at the register transfer level. At gate level you might add non-zero gate delays. However, models at gate level are much slower than models at register transfer level. Anyway, digital models at both levels discard usually the power consumption aspect to the point that even the power rail terminals Vdd and Gnd are not included in the block's interface declaration.

If you need information on power consumption, you would transform your gate-level netlist into a transistor-level SPICE netlist, which you would simulate in an analog simulator -- usually some variant of SPICE.

In conclusion, these analog aspects are out-of-scope of Logisim, which is in the first place a teaching tool for first year digital design classes.

hydrogenbomb1212, 2017-11-08:
ok, your right. i may want too mutch from a single program. ill use what i have for now. but the analog part is still something id like to see.
i am self learning in digotal design so i dont have a school or buisness to afford altera quartus for now. which would be the ideal solution.

closed 2017-11-08 by hydrogenbomb1212

LordDecapo, 2017-11-08:
Quartus Prime and Modelsim are free for small scale projects btw... some
functions limited, but enough to run tests on how a circuit will run and
perform on a given fpga