logisim-evolution/logisim-evolution #138: Crash on startup

piegamesde, 2017-11-06:
When I start the program, I get the following exception:

[image]

I am running the latest Windows 10, the latest Java 64bix and the latest version of logisim.

leadrien, 2017-11-06:
Hello,
It is the same than #131 and #133.
It has been fixed in be4ce639e177b880fe9297eebf29e843c2756cc0 but no .jar release has been published yet.
You can compile Logisim from the sources with ant, or wait for the next release.

closed 2017-11-06 by leadrien

piegamesde, 2017-11-06:
I did not know this was the same issue because one is named "(maxOS)" and both have a different stack trace.

When will the jar be released? This is a critical bug that completely breaks the program (tested on multiple devices) and thus makes it useless