logisim-evolution/logisim-evolution #142: Logisim Evolution forces the dGPU to be used

SamusAranX, 2018-01-08:
This is not necessarily an issue with Logisim Evolution itself, but rather an issue with Java, which, by default, enables the dGPU on MacBooks with both an integrated and a discrete GPU, wasting massive amounts of energy for basically nothing.
We use Logisim on our own laptops in school and this is a dealbreaker, as I won't get through the day with the dGPU running on full blast.

I checked Google for a solution and apparently, Java's default behavior can be disabled by creating an app bundle and setting NSSupportsAutomaticGraphicsSwitching to true in the bundle's Info.plist.
I lack the means to create an app bundle from Logisim's source code myself, but you providing such app bundles in the future would really help those of us using MacBooks.

maehne, 2018-01-10:
Thanks for reporting this issue! I can reproduce the issue on my MacBook Pro. The suggested fix works. I verified it by patching manually the Info.plist in my Logisim-evolution.app bundle:
```
<?xml version="1.0" ?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
 <dict>
  <key>LSMinimumSystemVersion</key>
  <string>10.9</string>
  <key>CFBundleDevelopmentRegion</key>
  <string>English</string>
  <key>CFBundleAllowMixedLocalizations</key>
  <true/>
  <key>CFBundleExecutable</key>
  <string>Logisim-evolution</string>
  <key>CFBundleIconFile</key>
  <string>Logisim-evolution.icns</string>
  <key>CFBundleIdentifier</key>
  <string>com.cburch.logisim</string>
  <key>CFBundleInfoDictionaryVersion</key>
  <string>6.0</string>
  <key>CFBundleName</key>
  <string>Logisim</string>
  <key>CFBundlePackageType</key>
  <string>APPL</string>
  <key>CFBundleShortVersionString</key>
  <string>2.14.2</string>
  <key>CFBundleSignature</key>
  <string>????</string>
  <!-- See http://developer.apple.com/library/mac/#releasenotes/General/SubmittingToMacAppStore/_index.html
       for list of AppStore categories -->
  <key>LSApplicationCategoryType</key>
  <string>public.app-category.education</string>
  <key>CFBundleVersion</key>
  <string>2.14.2</string>
  <key>NSHumanReadableCopyright</key>
  <string>Copyright (C) 2001--2017 Carl Burch, BFH, HEIG-VD, HEPIA, et al.</string>
  <key>CFBundleDocumentTypes</key>
  <array>
    <dict>
      <key>LSItemContentTypes</key>
      <array>
        <string>com.cburch.logisim.circ</string>
      </array>

      <key>CFBundleTypeName</key>
      <string>Logisim-evolution circuit file</string>

      <key>LSHandlerRank</key>
      <string>Owner</string>

      <key>CFBundleTypeRole</key>
      <string>Editor</string>

      <key>LSIsAppleDefaultForType</key>
      <true/>

    </dict>
  </array>

  <key>UTExportedTypeDeclarations</key>
  <array>
    <dict>
      <key>UTTypeIdentifier</key>
      <string>com.cburch.logisim.circ</string>

      <key>UTTypeDescription</key>
      <string>Logisim-evolution circuit file</string>
      <key>UTTypeConformsTo</key>
      <array>
          <string>public.data</string>
      </array>

      <key>UTTypeTagSpecification</key>
      <dict>

        <key>public.filename-extension</key>
        <array>
          <string>circ</string>
        </array>
        <key>public.mime-type</key>
        <array>
          <string>application-prs.cburch.logisim</string>
        </array>
      </dict>
    </dict>
  </array>

  <key>NSHighResolutionCapable</key>
  <string>true</string>
  <key>NSSupportsAutomaticGraphicsSwitching</key>
  <string>true</string>
 </dict>
</plist>
```
Adding the key ```NSSupportsAutomaticGraphicsSwitching``` through a simple addition to ```build.xml``` is not straight-forward. The whole ```Info.plist``` needs to be replaced, see [Customizing the Package Using Drop-In Resources](https://docs.oracle.com/javase/9/deploy/self-contained-application-packaging.htm#JSDPG593) in the JDK 9 documentation.

Fixing this issue depends on issue #135 being fixed.

alexandremalki, 2018-02-07:
@SamusAranX Did the commit  5cca9c5 solved the issue?

SamusAranX, 2018-02-24:
Sorry for the late reply, I only just got around to actually testing this.
I downloaded the latest release (2.14.4), but it still force-enables the discrete GPU.

If the fix hasn't made it into a release, I can't test it, as I neither have a Java IDE nor the knowledge to wrangle the code into a working .app package. :/

maehne, 2018-09-23:
To fix this issue, you will have to use Logisim-evolution packaged as a native app bundle for macOS, as only the latter includes the Info.plist, which prevents the dGPU to be used. I can reproduce the issue that dGPU gets used with the pure JAR, but it is not used when using the app bundle generated using the ```ant package``` on macOS. We should consider start releasing also the OS-specific Logisim-evolution packages for Windows (MSI installer), Linux (DEB and RPM package), and macOS (DMG archive).

closed 2018-10-20 by BFH-ktt1