logisim-evolution/logisim-evolution #145: Issue when comming back from sleep/locking

alexandremalki, 2018-02-13:
In fedora 27 | Kernel Linux 4.14.16 | Gnome 3.26.2:
Logisim rendering crash, when coming back from sleep/locking. The only way to recover from this is to restart Logisim. 
This is a known issue https://bugs.openjdk.java.net/browse/JDK-8147542 .

Solution/Workaround:
* Surround with try catch critical parts in java source code.
* Wait for Java to release new version of the jre.

*Attached, the crash report.*
[crash_report_log.txt](https://github.com/reds-heig/logisim-evolution/files/1718816/crash_report_log.txt)

maehne, 2018-06-14:
This should be fixed in JDK 9 and above as you can see from the referenced OpenJDK issue.

BFH-ktt1, 2020-04-05:
Due to long inactivity closing the issue, please reopen when the problem still persists.

closed 2020-04-05 by BFH-ktt1