logisim-evolution/logisim-evolution #147: Unifying version number changing

alexandremalki, 2018-02-13:
After introducing 5cca9c57 by @maehne the version number has to be modified each time in the file src/com/cburch/logisim/Main.java and the file Info.plist.

Please find a solution to unify that. So there is only one file to change. 

Thank you.

MarcinOrlowski, 2021-07-13:
@maehne Can you please change subject of this ticket to be more self explanatory please?

maehne, 2021-07-13:
@MarcinOrlowski: Done.

We still have two places to edit for updating the Logisim-evolution version number:

* `gradle.properties`
* `src/main/java/com/cburch/logisim/Main.java`

MarcinOrlowski, 2021-07-13:
I think we should generate java file with version number from Gradle so any version bump would be in one place. Will take a look.

MarcinOrlowski, 2021-09-26:
Closing as this is already addressed by #885

* #885

closed 2021-09-26 by MarcinOrlowski