logisim-evolution/logisim-evolution #157: Additional components request.

Nightquaker, 2018-04-08:
Hello there!
It would be quite nice if certain components from Logisim-ITA were integrated.
For example, the switches (normal ones, not dip switches) and digital oscilloscope.
And, if at all possible, the TTL library.

BFH-ktt1, 2018-07-31:
I contacted the developers of Logisim-ITA and await their respons.

BFH-ktt1, 2018-08-01:
The TTL-library of logisim-ITA has been merged into the scaling_resolution_wip. After my holidays I hope to continue merging the very nice features of logisim-ITA.

BFH-ktt1, 2018-08-14:
I have merged all logisim-ITA components like switches oscilloscope PLA-rom, etc. in a special lbuild-in library called logisim-ITA components in the scaling_resolution_wip branch. With this I close this issue; have fun and thanks a lot to the logisim-ITA crew!

closed 2018-08-14 by BFH-ktt1