logisim-evolution/logisim-evolution #158: Replace hard coded hardware device index number when creating Vivado download script

wittlage, 2018-05-17:
I would like to create a board description for Digilent ZYBO. Everything seems to be fine, but the created download script vivadoLoadBitStream.tcl contains a hard coded number that prevents downloading the successfully created bitstream.

[VivadoDownload.java](https://github.com/reds-heig/logisim-evolution/blob/master/src/com/bfh/logisim/download/VivadoDownload.java)

```
// load bitstream
String lindex = "[lindex [get_hw_devices] 0]";
```

IMHO the device index should not be hard coded but should use the JTAGPos from the boardInfo object.

wittlage, 2018-05-17:
`String JTAGPos = String.valueOf(boardInfo.fpga.getFpgaJTAGChainPosition());`
`String lindex = "[lindex [get_hw_devices] " + JTAGPos + "]";`

leadrien, 2018-05-17:
Thank you for this suggestion. I'll test and push the fix asap

leadrien, 2018-05-22:
Fixed in 394198ab6be69413d3a0e90a8318ccb5a5eb4b0e

closed 2018-05-22 by leadrien