logisim-evolution/logisim-evolution #163: TCL Component broken?

derbroti, 2018-07-02:
Hi,

I am trying to use the TCL generic component. 

First of all what does this line in the tcl_wrapper.tcl mean? It is nowhere else in the entire source.
https://github.com/reds-heig/logisim-evolution/blob/8749f5783e8aef4d83861230d1732f4ecf3593a8/resources/logisim/tcl/tcl_wrapper.tcl#L252

With it I get:
```
invalid command name "enableLogisim"
    while executing
"enableLogisim TRUE"
    (procedure "init_wrapper" line 7)
    invoked from within
"init_wrapper"
    (file "/tmp/logisim/tcl/tcl_wrapper.tcl" line 284)
```

Removing it does not help though:
```
[TclWrapperListenerThread-0] ERROR com.cburch.logisim.util.SocketClient - Cannot read from socket : Socket closed
[TclWrapperListenerThread-1] ERROR com.cburch.logisim.util.SocketClient - Cannot read from socket : Socket closed
```

I built the latest master and my TCL-VHDL port definition is most trivial:
```
library ieee;
use ieee.std_logic_1164.all;

entity TCL_Generic is
  port(
    val_i      : in  std_logic_vector(7 downto 0); -- input vector example
    val_o      : out std_logic_Vector(7 downto 0)  -- output vector example
  );
end TCL_Generic;
```

The tcl content file is the same trivial example from the docs containing just the refresh callback (which is never directly called in the source?! - a search for "refresh" gave no hits)

(even with an empty tcl file it does not work).
 
Any hints on what I am doing wrong?
-> I also tried to activate the VHDL simulator - which by itself works flawlessly but it did not help for the tcp component case.

Cheers!

derbroti, 2018-07-02:
Sorry for the mess, I found this line: 
https://github.com/reds-heig/logisim-evolution/blob/8749f5783e8aef4d83861230d1732f4ecf3593a8/src/com/cburch/logisim/std/tcl/TclComponentData.java#L118

and, if I am not getting this whole thing totally wrong: the tcl_wrapper.tcl is simply terminating -> the fileevent does not get called (at least in my version 8.6 of tclsh) -> a simple `vwait forever` at the end solves this but now restarting the simulator spawns a new tclsh session and never kills the old one -> but I don't see the stop or restart of TclWrapper ever called, or TCL_WRAPPER_ENDED evaluated for that matter

BFH-ktt1, 2019-01-23:
Is there some news on this issue?

MarcinOrlowski, 2021-07-13:
Bump? If this is not still the case, what are the implications?

maehne, 2021-07-13:
As far as I know, the TCL components are (or were?) almost exclusively at HEIG-VD in Yverdon. @reds-heig: Can you give some feedback?

BFH-ktt1, 2026-09-01:
Closing this issue as long time no action

closed 2026-09-01 by BFH-ktt1