logisim-evolution/logisim-evolution #166: [Issue] Exception in thread "main" java.awt.HeadlessException when sshed on a remote server

ThaumicMekanism, 2018-07-13:
Hello, I am having this error when I try to run logisim evolution on a remote server in tty table mode:

> Exception in thread "main" java.awt.HeadlessException:
> No X11 DISPLAY variable was set, but this program performed an operation which requires it.
>         at java.awt.GraphicsEnvironment.checkHeadless(GraphicsEnvironment.java:204)
>         at java.awt.Window.<init>(Window.java:536)
>         at java.awt.Frame.<init>(Frame.java:420)
>         at java.awt.Frame.<init>(Frame.java:385)
>         at javax.swing.SwingUtilities$SharedOwnerFrame.<init>(SwingUtilities.java:1758)
>         at javax.swing.SwingUtilities.getSharedOwnerFrame(SwingUtilities.java:1833)
>         at javax.swing.JOptionPane.getRootFrame(JOptionPane.java:1696)
>         at javax.swing.JOptionPane.showOptionDialog(JOptionPane.java:863)
>         at javax.swing.JOptionPane.showMessageDialog(JOptionPane.java:666)
>         at javax.swing.JOptionPane.showMessageDialog(JOptionPane.java:637)
>         at javax.swing.JOptionPane.showMessageDialog(JOptionPane.java:608)
>         at com.cburch.logisim.Main.main(Unknown Source)

We have tried to forward the display but our server does not allow for it and no one has access to doing that. I had then tried older versions of logisim evolution (since it did work on the original logisim) and discovered that the latest version I could get it to work with was 2.13.10 and it does NOT work with 2.13.14. (Note: I had tried to compile the version in between them but was having compilation issues and did not want to spend too much time doing that). Do you have any idea what changed to make it so that tty table is now requiring the graphical interface to load? Like was this intentional or is this just a mistake?

Thanks, Hopefully, you can take a look at this quickly since we want to use this with a project but are having issues where students cannot test their projects because of this issue. (They need to test on the remote server but they still build it on their local machine)

maehne, 2018-07-13:
> We have tried to forward the display but our server does not allow for it and no one has access to doing that.

Just to make sure: You tried to connect to the server via ```ssh -X <server>``` and ```ssh -Y <server``` to forward the display? Sometimes SSH servers only accept one of the two variants.

Version 2.13.14 was released in December 2015. Have you tried with the lates released version of logisim-evolution, i.e., version [2.14.6](https://github.com/reds-heig/logisim-evolution/releases)?

I am not familiar enough with the source to make a good guess, why the tty table mode has stopped working when no DISPLAY variable is present. Hopefully, someone of the main developers can have a look at it, but they might be well occupied with tasks linked to the end of the spring term.

ThaumicMekanism, 2018-07-13:
Yes, I have tried both -X and -Y and neither of them work. I also tried to set the display but that also did not help. I have tried the newest version. The reason I was saying those versions was because it works fine up to 2.13.10 and does not work with versions after 2.13.14 including the latest version. I had tried all of the versions to pinpoint which version caused the issue so it could possibly help with debug. Also we even tried using the forced headless mode in java but that also does not help.

maehne, 2018-07-13:
Thanks for the clarification! The version span 2.13.10 to 2.13.14 definitely should help to localise the change introduced between 2015-08-28 and 2015-12-07, which causes the problem.

@BFH-ktt1: Do you have maybe an idea?

ThaumicMekanism, 2018-07-13:
I fixed one issue where it was erroring the correct error message (because in main, you are trying to write to a gui when it is not open causing another error). Here is the new one:

> java.lang.ExceptionInInitializerError
>         at com.cburch.logisim.util.Icons.getIcon(Icons.java:52)
>         at com.cburch.logisim.tools.SelectTool.<clinit>(SelectTool.java:134)
>         at com.cburch.logisim.std.base.Base.<init>(Base.java:50)
>         at com.cburch.logisim.std.Builtin.<init>(Builtin.java:55)
>         at com.cburch.logisim.file.Loader.<init>(Loader.java:127)
>         at com.cburch.logisim.gui.start.TtyInterface.run(TtyInterface.java:251)
>         at com.cburch.logisim.gui.start.Startup.run(Startup.java:801)
>         at com.cburch.logisim.Main.main(Main.java:71)
> Caused by: java.awt.HeadlessException:
> No X11 DISPLAY variable was set, but this program performed an operation which requires it.
>         at sun.awt.HeadlessToolkit.getScreenSize(HeadlessToolkit.java:284)
>         at com.cburch.logisim.prefs.AppPreferences.<clinit>(AppPreferences.java:531)
>         ... 8 more

So what I am guessing is that the main file makes sure that it does not load the interface but it still is trying to load parts of it anyway after.

ThaumicMekanism, 2018-07-13:
I made a patch which solves this issue and should not affect anything else. Just made a pull request.

reds-heig, 2018-07-30:
Fixed by @ThaumicMekanism , thanks a lot for your help! :)

closed 2018-07-30 by reds-heig