logisim-evolution/logisim-evolution #171: Problem with UTF8 display under windows.

mbaillif, 2018-08-10:
With the autoex of the .jar file the display UTF8 is not recognized under windows (test with W10 and Java 8 and version 2.14.6)
this gives a nasty effect with accented letters.

I notice that the files of definition of texts are sometimes in ANSI and sometimes in UTF8
Which is the future vision for this coding ?, It seems to me that UTF stands out

temporary solution:
create a shortcut and place this command in the target field:  javaw.exe" -Dfile.encoding=UTF8 -jar "E:\Users\Public\LogisimEvolution\logisim-evolution.jar"

or
create a .bat file with this line:
start javaw -Dfile.encoding=UTF8 -jar E:\Users\Public\LogisimEvolutionlogisim-evolution.jar

Change the file path of the .jar file as needed

maehne, 2018-09-23:
Thanks for reporting this inconsistency in the Logisim-evolution source files! They are testimony of the quite diverse development platforms of its contributors. I went through the source files and hopefully fixed all encodings to be now consistent (UTF-8 with Unix-style line endings for most cases and even pure ASCII for *.properties files). The result is in my [unify-encodings](https://github.com/maehne/logisim-evolution/tree/unify-encodings) branch. I opened PR #177 to get it merged into the scaling-resolution-wip branch, from which it should get soon also into master. In the meantime, would you mind testing the branch on your platform to see whether it fixes your issue completely?

mbaillif, 2018-09-25:
(Version 2.14.6)
Thank you for your files<br>
After the tests and a little analysis of the situation I observe two behaviors in two part of Logisim.

<p><b>A) In the software itself with the .properties files</b><br>
These files are still in ISO-8859-1 with esc tags of type \uxxxx for languages like Greek or Russian.
<table>
<tr>
<td></td><td>UTF8</td><td>ISO</td>
</tr>
<tr>
<td>MAC OS</td><td>bad</td><td>OK</td>
</tr>
<tr>
<td>WIN 10</td><td>bad</td><td>OK</td>
</tr>
<tr>
<td>Debian</td><td>bad</td><td>OK</td>
</tr>
</table></p>

<p><b>B) In the help section of the program (java help)</b><br>
Here we have HTML files and a tag  &lt;meta charset="UTF-8"&gt; or  &lt;meta charset="ISO-8859-1"&gt; <br>
UTF8 * = with option java -Dfile.encoding=UTF8
<table>
<tr>
<td></td><td>UTF8</td><td>UTF8 *</td><td>ISO</td>
</tr>
<tr>
<td>MAC</td><td>bad</td><td>OK</td><td>bad</td>
</tr>
<tr>
<td>WIN 10</td><td>bad</td><td>OK</td><td>OK</td>
</tr>
<tr>
<td>Debian</td><td>OK</td><td>OK</td><td>bad</td>
</tr>
</table>
</p>
<p>
<b>My conclusions for now</b><br>
- properties files must remain in ISO never in UTF<br>
- Help html files must be UTF and always launch with the option <tt>java -Dfile.encoding = UTF8 -jar logisim-evolution.jar</tt><br>
Maybe it's possible to do it in the configuration settings. I do not have enough skills for that
</p>

maehne, 2018-09-25:
Version 2.14.6 does not contain the fixes from my [unify-encoding](
https://github.com/maehne/logisim-evolution/tree/unify-encoding) branch. If you cannot build logisim-evolution yourself using JDK >=8, you will have to wait for the next binary release of Logisim-evolution. I mostly agree with your above analysis and therefore my PR #177 should address your issue. With my patches, I could not find occurences of wrongly displayed accented characters in the GUI (using the ACII encoded *.properties files with Unicode escapes) nor the help files, which should now all use UTF-8. I was not able to find a file declaring ```<meta charset="ISO-8859-1"> ``` nor any other file containing "ISO-8859-1". Though, UTF-8 encoded HTML files seem to display just fine even without specifying ```-Dfile.encoding = UTF8``` when launching ```logisim-evolution.jar```. Could you maybe point me to a specific HTML file and GUI window, where you observed display problems? Best would be if you would post a suitable screenshot here and say which language is affected.

maehne, 2018-09-25:
I realise now that this issue seems to be linked to your considerable effort (issue #172, [mbaillif/logisim-evolution](
https://github.com/mbaillif/logisim-evolution/commits/master)) to improve the French translation of Logisim-evolution, which you based on our release 3.14.6. Thanks for this considerable effort! This work will need to get rebased on our development HEAD and potential conflicts will need to get resolved (to avoid to reintroduce encoding inconsistencies). As @BFH-ktt1 wrote, we will try to get to this once our work load it permits (the start of the term is always a bit rough).

maehne, 2018-09-25:
I realise now that this issue seems to be linked to your considerable effort (issue #172, [mbaillif/logisim-evolution](
https://github.com/mbaillif/logisim-evolution/commits/master)) to improve the French translation of Logisim-evolution, which you based on our release 3.14.6. Thanks for this considerable effort! This work will need to get rebased on our development HEAD and potential conflicts will need to get resolved (to avoid to reintroduce encoding inconsistencies). As @BFH-ktt1 wrote, we will try to get to this once our work load has dropped (the first weeks of a new teaching term are always a bit rough).

mbaillif, 2018-10-12:
Actually I undertook the translation of the French version. I will make a ticket especially on this point.

Regarding the accented letters, I think I found the solution in a more rigorous formatting HTML header.
<pre>
&lt;!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 4.0//EN"&gt;
&lt;html&gt;
&lt;head&gt;
&lt;META http-equiv="content-type" content="text/html; charset=UTF-8"&gt;
&lt;META http-equiv="Content-Language" content="fr"&gt;
...
</pre>

maehne, 2018-10-15:
Thanks for looking further into the encoding issue. It seems to be Windows-specific, as Linux and macOS nowadays default to UTF-8. I built Logisim-evolution for the different supported platforms from @BFH-ktt1 development head, which I have made available here:

https://drive.switch.ch/index.php/s/KfMIaQbH7dxdI9l

Could you please test the JAR and MSI installer on Windows? In the Logisim-evolution GUI, I have not noticed any problems with accented letters. Though, HTML display in the French manuals seems still pose a problem. Interestingly, the German manuals seem to display accented letters (i.e., the umlauts ä, ö, ü) just fine. So maybe, you could look at what differs between those two manual versions. Enforcing the correct encoding either through a more rigorous HTML header or some measure in the HTML viewer would seem a good way to fix the problem.

mbaillif, 2018-10-28:
I will do it.
Html and install on
Windows Mac and linux ubuntu

mbaillif, 2018-11-02:
Windows: Installer
The installation went very well. The logisim updates work.Tested on Windows 10 / Windows 7 (New installation) The .jar also works without installation if Java is preinstalled

maehne, 2018-11-02:
Thanks for this feedback! Can you confirm that accented letters are displayed correctly in the GUI, but not in the help files?

mbaillif, 2018-11-02:
Mac (sierra): The installation works.
Linux debian: The installation works.

maehne, 2018-11-02:
Thanks for this feedback! Do you observe on Mac or Linux problems with accented letters in the GUI or help files?

mbaillif, 2018-11-02:
Après les testes je pens eque nous pouvons fermer ce ticket. C'est bien les hedaer HTTP qui font la différence.
Je l'ai corrigé dans l'ensemble des fichiers et dans toutes les langues.

mbaillif, 2018-11-02:
After the tests I think we can close this ticket. It's the HTTP hedaer that makes the difference.I corrected it in all files and in all languages.

mbaillif, 2018-11-02:
A small note on the components "pin" The outputs no longer show in color their state. 
This will be an advantage if these representations keep the same width as two squares (in 1 bit wide) in north-south orientation.

maehne, 2018-11-04:
> A small note on the components "pin" The outputs no longer show in color their state.
> This will be an advantage if these representations keep the same width as two squares (in 1 bit wide) in north-south orientation.

Thanks for this additional feedback! Could you please open a separate issue for this issue? Otherwise, it might get easily lost! @BFH-ktt1 modified the look of the component pins recently. So he should be able to easily fix it.

closed 2018-11-04 by mbaillif