logisim-evolution/logisim-evolution #172: Update of LPG in v3

mbaillif, 2018-08-16:
In the fork mbaillif branch test
I updated the text of the GPL (gpl.html) in its version V3 in accordance with the conventions
I also modify the text (fr / en ) of about (index.html) to match the recent description.
Can you check if this fits the idea of the project?
before going further

BFH-ktt1, 2018-09-18:
Dear mbaillif,
Thanks for the work, I will look soon into your modifs

mbaillif, 2018-09-22:
The most successful version of the translation is in the branch
mbaillif / logisim-evolution master.

Everything has been written to fit the sun javahelp library.
There is still work but the doc fr is presentable.

The guide section is almost complete, the test vectors page, and the fpga related parts are still missing.

I already have the figures for the English version.

BFH-ktt1, 2019-01-23:
I am not sure if with the latest pull requests we integrated all. Are there open tasks or can we close this issue?

mbaillif, 2019-01-24:
Everything is in order, can be closed

closed 2019-01-24 by mbaillif