logisim-evolution/logisim-evolution #179: Circuit Randomly Breaks

SpaceboyRoss01, 2018-10-05:
I'm building a CPU and when I had placed the CPU circuit onto the motherboard circuit, the CPU showed a red outline. I checked the other files and my CPU shows a red outline around the 16-Bit ALU, then I checked the 16-Bit ALU and the 1-Bit ALU's have a red outline. I don't know what's going on and how to fix it.

# 1-Bit ALU
[image]

# 16-Bit ALU 
[image]

# CPU
[image]

# Motherboard
[image]

BFH-ktt1, 2018-10-07:
You have a combinational loop in your carry-chain that causes oscillation. You should put somewhere in your carry-chain a flipflop to break this oscillation.

BFH-ktt1, 2018-10-07:
In addition you have a short-circuit in the input less and the output of the adder cout. This will also cause in certain conditions a problem.

closed 2018-10-07 by BFH-ktt1