logisim-evolution/logisim-evolution #186: V2.14.7  The outputs no longer show in color their state

mbaillif, 2018-11-04:
A small note on the components "pin" The outputs no longer show in color their state.
This will be an advantage if these representations keep the same width as two squares (in 1 bit wide) in north-south orientation (reverse compatibility).

mbaillif, 2018-11-29:
Precision
[image]

BFH-ktt1, 2018-12-18:
Please take a look at the latest version. The sizes have been fixed for the single bit. My action plan is to move even further and keep the size of the input-symbols equal in each direction by rotation the bit-vectors in case of north and south orientation. As soon as I have this realized I will post a comment.

BFH-ktt1, 2019-02-22:
This should be fixed in the latest version, closing the issue, please feel free to reopen if you think there is still an issue.

closed 2019-02-22 by BFH-ktt1