logisim-evolution/logisim-evolution #189: ROM content won't change after editing it

Censacrof, 2018-11-21:
Im using logisim-evolution 2.14.6

### Steps to reproduce:
- create a new circuit
- place a ROM in it
- edit its content

### What happens then?
The ROM will behave as if nothing changed. I noticed that if you try to edit it again the hex editor will show the prover values even if the ROM doesn't seem to care.

### Workaround
The only way i've found to use the ROM is to edit its content after selecting it in the circuits tab before placing it in a circuit.

BFH-ktt1, 2018-11-22:
I tried to reproduce this issue on the stable version 2.14.6 (latest release) and scaling_resolution_wip (2.14.7). In both versions I cannot reproduce the behavior. What did I test:
1) I created a new circuit.
2) I placed a ROM in it, an input-pin on the address-lines, and an output pin on the data-lines.
3) With the poke-tool I selected multiple memory-positions and changed their value.
4) I selected the proper addresses with the poke-tool and observed the output value; they were all okay.
5) I repeated the above procedure several times. All worked as it should.
6) Opened the hex-editor, it also showed the correct values.
Than I tried following:
1) I created a new circuit.
2) I placed a ROM in it, an input-pin on the address-lines, and an output pin on the data-lines.
3) I opened the hex editor and changed the contents of the ROM.
4) I selected the proper addresses with the poke-tool and observed the output value; they were all okay.
5) I repeated the above procedure several times. All worked as it should.
6) Opened the hex-editor, it also showed the correct values.
Finally I tried a combination of the above two methods and could not observe any strange behavior.

Could you please provide a more detailed description (or for example a logisim file) that helps me debug your observations and find a possible bug in the ROMS?
Thanks in advance!

Censacrof, 2018-11-22:
This is the incriminated file.

[RetiLogiche.zip](https://github.com/reds-heig/logisim-evolution/files/2608715/RetiLogiche.zip)

## This is getting weirder
As soon as i saw your reply i tried to reproduce the bug again. But this time it worked. So, i tried change the 'Address Bit Width' and the 'Data Bit Width' at random (afte the ROM is placed) and *eventually the bug occurred again*. Now i'm really confused. 

Also i didn't mention this in the previus comment but the project was Imported from the standard Logisim, i don't know if this information is useful.

Anyway thanks for your time, don't esitate to ask me for more clarifications.

BFH-ktt1, 2018-12-18:
Do you have some new on the issue, is it resolved? Can we close?

closed 2018-12-22 by BFH-ktt1

Codyl, 2020-06-05:
If anyone has this issue, make sure you have simulation enabled in the simulate drop down. That might fix the issue if it is simple. Otherwise, it may not let you edit the rom.

maehne, 2020-06-05:
@Codyl: Does it mean that this issue should be reopened because it is still present in the latest release 3.3.4 of Logisim-evolution? Do you have a test case / sequence of steps to reproduce reliably this issue?

wwachowski, 2023-10-17:
To resolve, Reset Simulation (Ctrl + R)

maehne, 2023-10-17:
@wwachowski: It is not clear whether you also observe this issue. If yes, please state the exact version of Logisim-evolution, the JDK version and OS platform you are using. Then, please provide a sample circuit (as ZIP archive attachment to a reply to this issue) as well as precise steps how to reproduce what you are observing.

wwachowski, 2023-10-17:
@maehne 
I am sorry for the confusion.
I did not pay enough attention to see that this thread is devoted to Logisim-evolution, not the primal Logisim.
I believe that Codyl also might have had original Logisim in mind

maehne, 2023-10-22:
@wwachowski: As the development of the original Logisim was stopped years ago. I recommend you considering to switch to a maintained fork. The longer you wait, the harder it might become to migrate old Logisim circuits to work in one of the maintained forks.