logisim-evolution/logisim-evolution #191: too slow to start logisim-evolution.jar on linux

recolic, 2018-11-28:
I downloaded the stable release 2.14.6 [here](https://github.com/reds-heig/logisim-evolution/releases). While running `java -jar logisim-evolution.jar`, it hangs several minutes before the window appears.

My system information:
```
➜  program uname -a
Linux RECOLICPC 4.19.2-arch1-1-ARCH #1 SMP PREEMPT Tue Nov 13 21:16:19 UTC 2018 x86_64 GNU/Linux
```

I tried to use java 11 and java 8. It doesn't help. Here's the log and java version:
```
➜  program time java -jar logisim-evolution.jar 
7.33user 0.45system 2:16.40elapsed 5%CPU (0avgtext+0avgdata 216368maxresident)k
32inputs+288outputs (0major+52539minor)pagefaults 0swaps
# close the program once the main window appears
➜  program java -version
openjdk version "11.0.1" 2018-10-16
OpenJDK Runtime Environment (build 11.0.1+13)
OpenJDK 64-Bit Server VM (build 11.0.1+13, mixed mode)
➜  program sudo archlinux-java status
Available Java environments:
  java-11-openjdk (default)
  java-7-openjdk
  java-8-openjdk
➜  program sudo archlinux-java set java-8-openjdk
➜  program time java -jar logisim-evolution.jar 
4.70user 0.38system 2:13.51elapsed 3%CPU (0avgtext+0avgdata 188164maxresident)k
0inputs+8outputs (0major+45711minor)pagefaults 0swaps
# close the program once the main window appears
➜  program java -version
openjdk version "1.8.0_192"
OpenJDK Runtime Environment (build 1.8.0_192-b26)
OpenJDK 64-Bit Server VM (build 25.192-b26, mixed mode)
```

maehne, 2018-11-29:
Thanks for the report! I have not yet observed such slow start up times on my Debian GNU/Linux installation nor on macOS nor on Windows. However, you seem to observe the issue right after installing the JDK. It could be that at first start of Logisim-evolution right after the installation of the JDK, the Java runtime has to perform a lot of just-in-time (JIT) compilation, which would explain such a slow start up. Have you tried whether you observe such slow start up times if you run the same Logisim-evolution JAR again? Do you observe such slow start up performance with other Java-based programs?

recolic, 2018-12-07:
The start up is still slow if I try again. I'm using java8 to play minecraft, using java11 to do my java development. I had never observe such slow start up performance with other java program...

It can be reproduced stably..

maehne, 2018-12-07:
May this be related to your internet connection? Logisim-evolution does check at each startup whether there's a newer version available. This could lead to a timeout in case of slow internet connection. Have you observed the slow startup times systematically independent of the network you are on or even no internet connection?

recolic, 2018-12-07:
Please wait. I'm testing it now...

BFH-ktt1, 2018-12-08:
People may also have this effect when using openJDK11. I had today also very slow start-up performance using openjdk installed in opensuse tumbleweed. Switching to oracles JDK fixed the problem.

recolic, 2018-12-09:
My problem is there network io. but I think network operations should never block ui thread, isn't it?

we should make it async.

maehne, 2018-12-09:
The URL checked at startup whether updates are available is:

[http://reds-data.heig-vd.ch/logisim-evolution/logisim_evolution_version.xml](http://reds-data.heig-vd.ch/logisim-evolution/logisim_evolution_version.xml)

You can check using your web browser, whether accessing it is the source of the problem. I agree with your statement that the update check should not block the UI thread. A PR to fix it is welcome!

In the meantime, you can disable the update check at compile time by editing `src/com/cburch/logisim/Main.java` and setting the boolean flag `UPDATE` in line 102 to `false` or when starting the JAR by passing the option `-noupdates` to `logisim-evolution.jar`.

recolic, 2018-12-09:
I think `-noupdates` is a good solution... Now everything is OK.

BFH-ktt1, 2018-12-18:
see issue #199

BFH-ktt1, 2018-12-18:
Closing it for now, please feel free to reopen or to move to/comment on issue #199

closed 2018-12-18 by BFH-ktt1