logisim-evolution/logisim-evolution #193: Problem with the test vectors.

mbaillif, 2018-11-28:
Problem with the test vectors.
In some cases, the test does not work properly.
Example at the top, the circuit with its vector file and the correct result.

At the bottom the introduction of three "not" doors will generate errors.
It can be seen that in the "Ob" column these errors are not detected.
All columns should be red.
Apparently when there is an error in the first column the errors of the second column are not detected.

[image]

maehne, 2018-11-29:
Thanks for the report! Could you please provide the corresponding Logisim circuit file(s) to facilitate reproducing the error?

mbaillif, 2018-12-01:
originally I observed the problem on a more complex circuit.
I have reduced this circuit to its simplest expression to test this problem.
There remain only three inputs connected to three outputs with the test vectors file as shown in the figure.
I added "Not" gates so that the outputs are in error with respect to the vector file.
I could also edit the vector file without modifying the schema.
This problem arises as well for 1/0 errors as for E or X

BFH-ktt1, 2018-12-23:
This was really a bad one..... The test vector would only show the first and the last error in the colums due to a malformed data structure (add would always override an internal value :-<). Should be fixed by my latest pull request

BFH-ktt1, 2018-12-23:
Please re-open in case other problems arrise.

commit db5621fa (2020-01-09): Due to a restriction in the FailException data structure only 2 error
could be stored. In case of more errors only the first and the last
would be stored. This fixes #193
closed 2018-12-23 by BFH-ktt1