logisim-evolution/logisim-evolution #194: Trying to use a splitter to ''bundle'' 36 bits together

claudiojar, 2018-11-29:
Hello everyone,
It's my first time posting on this forum so I don't know if any of you can help me ! 

My problem is this : I'm trying to create a bus where I carry a 42 bit long value, but I do not know how to do it. I basically need to carry the information to use 6 7-segment displays (without the dot) along a single wire. I tried to combine the 7 values for each display using a splitter into a single wire (for each display) so I got 6 wires each carrying a 7 bit value. Then I tried using another splitter to combine these 6 7-bit values into a single wire carrying a 42 bit value, but the splitter only allows me to go up to a 32 bit value (Bit Width In has a max value of 32 bit). 

Can anyone help me ? Does anyone have a better way to do this ? 

Thank You !!!

maehne, 2018-11-30:
Unfortunately, bus widths are limited to 32 bit in Logisim rooted in some early implementation choices. Therefore, it is difficult to lift this particular restriction, but PRs to address the issue are always welcome.

For the meantime, I would suggest that you use 6 port à 7 bits to connect to your 7 segment displays.

claudiojar, 2018-11-30:
Thank you so much @maehne !

closed 2018-11-30 by claudiojar

mkayaalp, 2020-01-20:
See #327. It increases the limit to 64 bits.