logisim-evolution/logisim-evolution #196: Switch to Gradle

Caellian, 2018-12-07:
This project should really start switching to some newer project management standards.
Most developers have stopped using Ant since Maven came out and some of them switched to Gradle.
Gradle would help with project organisation a great deal, simplify work on project with both Eclipse and IDEA.
I would be willing to write the build.gradle file and configure the project to use it if you don't insist on using Ant (which in my opinion no one should considering how much Maven and Gradle bring to the table).

maehne, 2018-12-07:
Thanks for your offer to contribute in form of migrating logisim-evolution to a more modern build tool! The timing is perfect: Among the main contributors, we reached the same conclusion that to further evolve the code base, we need to modernise the build system. `gradle` seems to us the good way. So please, feel free to make a PR with the necessary changes and we will review and very likely merge it!

Caellian, 2018-12-11:
I just started switching to Gradle. In order to follow latest conventions and work with gradle without 'hacking' the build system to work properly with current project structure here's what I need to do:

- move sources from src/ to src/main/java/
- move resource dirs (resources, javax, ...) to src/main/resources/
- unpack resources from resources directory and update all references from within code
- write tasks for package generation
- update README.md to describe usage with the new build system

The problem is that any commits made after I fork the package and restructure the project but merged in before this commit will (I'm 95% certain) cause conflicts. Also, anyone working on current project structure will have to pull my commit (and manually reapply all their changes to code (copy-paste)) in order to continue and commit their work.

I'm suggesting that instead of one big commit I make several smaller ones in order to reduce the effect of chaos I described above.

While entire sources don't have to be moved because gradle allows changing the path of source location, I highly suggest doing so at one point in way I described before (src/main/java) because that would make writing tests, managing resources and writing parts of code in different JVM languages (e.g. Kotlin) much easier. Instead of handling each resource directory separately and modifying gradle build file every time, everything would be handled automatically and imported into IDEs automatically.

I'd like you to voice your opinions and concerns before I start doing higher-level refactoring, until then I'll start implementing gradle to work properly with the current project stucture.

leadrien, 2018-12-11:
Hello,

Thank you for your contribution!

Since we decided to move to Gradle, we have to create the correct structure to respect its standards.
So the structure you described is perfect.

I think it's best to have a single Pull request with the complete new structure. And if there are some commit in between we will take care of adapting them afterwards.

It depends on how long it will take you to achieve this migration. Do you have any estimation?

Caellian, 2018-12-11:
I'll have it done by the end of today.

BFH-ktt1, 2018-12-18:
Thanks a lot for the great work! As master is now on gradle I close this issue

closed 2018-12-18 by BFH-ktt1