logisim-evolution/logisim-evolution #198: Bug with unfound pin.

LordDecapo, 2018-12-18:
I just recently downloaded the new v2.15.0 release. I went to download the VHDL of my circuit and it returned an error:
`*****FATAL***** BUG: Unable to find pin in sub-circuit ==> com/bfh/logisim/designrulecheck/Netlist:1117
`
NOTE: This isnt 100% reproducible on my system, I clicked the "Download" button multiple times, and about 1 in 20 clicks will work fine, _sometimes_. This last attempt to get it to download, it wouldnt work regardless of how many times I clicked the button.

I have used the last release (2.14.6) to download my VHDL from the same circuit with no issue. Just tried again with the same .circ that was giving me the issues in 2.15.0. 

I looked into the Netlist.class and saw this

 ```
if (SubPortIndex < 0) {
        Reporter.AddFatalError("BUG:  Unable to find pin in sub-circuit\n ==> " + 
          getClass().getName().replaceAll("\\.", "/") + ":" + Thread.currentThread().getStackTrace()[2].getLineNumber() + "\n");
```

Looks like the SubPortIndex is underflowing. I have pretty much 0 experience with the dev side of Logisim Evolution, but I hope the info I did give, is of use.

leadrien, 2018-12-18:
Can you post the .circ that causes the problem?
What board are you targeting?

LordDecapo, 2018-12-18:
Well, I just got home and tried to recreate it on my desktop... but it doesn't happen on my desktop, only my laptop.
Sadly, I can not send you the .circ, as it's not mine to share. :/ I will try and get the bug to happen with a different circ on my laptop, or with something on my desktop. Will post here when I can :)
Thanks for getting back with me quickly!

BFH-ktt1, 2018-12-18:
Could you also please check which java version you are running on your laptop and which one on your desktop? (java --version)

BFH-ktt1, 2018-12-18:
In this case we need a test circuit as I tried several (also complex) circuits that did not show any hick-up. If you could provide us with a test circuit that reproduces your observation I can start to debug. By the way, the targeted board is for this error of no importance it is related to the generation of the net-list where probably there exist multiple instances of a sub-circuit object of which, in the different stages of the generation, the "wrong" ones are used. This could be related to a bug that was present in the "merge" function in v2.14.6, which was fixed by me recently.

LordDecapo, 2018-12-18:
the target board was a DE-0 Nano. 
Currently working in the board editor to add more ports to it. Was running a test download when the bug occured. Will try and make a test circuit to recreate this bug.

BFH-ktt1, 2019-01-23:
Is there some update on this issue?

BFH-ktt1, 2020-04-05:
The problem should be solved in my latest pushes (it was related to "complex"-splitter connections). Closing the issue. Please reopen if the problem persists.

closed 2020-04-05 by BFH-ktt1