logisim-evolution/logisim-evolution #199: Default values of logic analyzer and software updates

BFH-ktt1, 2018-12-18:
For a while now the default values of the logic analyzer was on false and auto-update on true. For good reasons these values were chosen this way; however, it can cause problems.
I opt for following solution:
1) We add a gradle target that disables the logic analyzer for the jar-build. The default value will than be "on" so that the official releases have the analyzer enabled.
2) We add a construct that asks the user when logisim is for the first time run on a machine if the new-version check should be enabled. If so each time logisim starts the check will be performed, if not no check will be done. This setting should be also available in the preferences such that it can be enabled/disabled when the user wants.
As I am not a great star in gradle I could use some help on 1). Good ideas how to implement 2) are also welcomed.

maehne, 2018-12-19:
@BFH-ktt1: I agree with you that it would be nice to make both settings easily configurable by passing suitable options to the new `gradle` build system at compile time.

@Caellian: Would you maybe willing to support us in this task as we just start to get familiar with `gradle`?

Caellian, 2018-12-31:
https://stackoverflow.com/questions/51937034/gradle-how-to-change-version-number-in-source-code
That's the way to go.

So something along the lines of:
`compileJava { 
    doFirst {
        def destination = new File("${projectDir}/src/")
        def origin = new File("${projectDir}/template/")

        // TODO: Get all "**/*" files using grep or something
        // TODO: Iterate over each file in origin (template) directory:
        templateFile.withWriter { def writer ->
            templateBuildInfo.eachLine { def line ->
                def newLine = line.replace("@VERSION@", version)
                                  .replace("@LOG_ANALIZER_ENABLED@", logicAnalizerEnabled)
                writer.write(newLine + "\n");
            }
        }
    }
}`
and then create a new template source set which is included in src source set making the development seamless, finding configured files easily, easily adding configuration to new files by simply draging them into another source set. This means that we can get automatic version insertion on compilation going.

I'm a bit busy now, but I might be snug in some time to write that sometime soon. Shouldn't be too hard and would add amazing configurability to sources. Was just thinking of migrating my own projects to that format. And it should also be backwards compatible unlike my previous commit.

Caellian, 2019-11-30:
I started working on this issue locally. I'm almost done. I updated RSynaxTextArea to 4.0.3 which broke the code because the jar was manually patched. Patching a library jars is bad because it makes it more complicated to update dependencies.
I decompiled the library jar currently included by the repository and noticed you're using JFlex to generate the VHDLTokenMarker class.

Can you provide the JFlex source file you're using to generate VHDLTokenMarker? I'd like to include the file in the project and generate the source file using gradle which would allow the community to update the synax in case of possible bugs.

Another change I noticed in repository history is that you removed syntax highlighting for unused languages 3 years ago. I'll create an issue on RSynTA repository asking the developer to separate languages into separate packages which can then be installed individually if needed.

Regarding changes to the build system - I added a new source set containing files which need to be processed (Files containing version information and default values). Adding build-time configurable values should be really straightforward now.

Sorry it took me a year to respond back, I forgot about this issue.

BFH-ktt1, 2019-11-30:
Dear Caellian,

Thanks a lot for your work. To your questions/remarks:
The flex file should be: com/cburch/logisim/vhdl/syntax/VhdlSyntax.flex. As this part of the code was created by Kevin Walsh (I merged his features  of his logisim Holy cross edition in evolution) he might be able to to give some more detailed info (also concerning the RSyntax).
It was still on my todo list to remove the RSyntaxTextarea jar and replace it by a direct pull, so I am very pleased to hear that you already looked into this part, thanks.

Caellian, 2019-12-01:
Found it in his repo. I will postpone removing RSTA jar as it's a very intricate problem, I'll create a separate issue describing it.

As for this issue, a pull-request will be coming soon. Maybe today.

Caellian, 2019-12-01:
Regarding 2nd part. I suggest you create a file in:
- Windows: %appdata%
- Unix: ~/.local/share/
Containing configuration for the program. Not sure if you already have configuration files, but that is the best way of doing this.

I suggest you use YAML format if GSON (JSON) or a similar library isn't already used by the project, this would make editing config files using a text editor very simple and less error prone. Most libraries allow direct serialization of objects so all this should end up being is ~3 lines of code for loading and saving the configuration object, and the object should be a very straightforward class containing only values configurable through your config file.

This is [very trivial](https://dzone.com/articles/read-yaml-in-java-with-jackson) to do and it's a great start for someone new to the project. [GSON version](https://howtodoinjava.com/gson/google-gson-tutorial/).
Jackson also supports json out of the box but I personally prefer YAML.

[TOML](https://medium.com/swlh/configure-your-java-program-with-toml-file-9f779a6a3de6) is also a good option for simple configurations but gets tedious to use for more complicated data structures.
In case you wish to store other data (apart from configuration) using one of these common formats, I suggest against using TOML and uncompressed XML.

maehne, 2019-12-01:
The FPGA Commander already stores the information needed for each development board in XML format. XML is also used in many other places. Maybe, it would be good to decide on a single configuration format. Personally, I find JSON, YAML, or TOML easier much more readable for a human than XML.

Caellian, 2019-12-02:
JSON has the smallest memory footprint with GSON taking up ~240KB, YAML (Jackson + YAML format support) takes up ~390KB.
Whichever is chosen I suggest FPGA Commander should also use the same format and if necessary files can be further compressed using zip format support which is part of JRE. XML apart from being hard to read is also the bulkiest of them as it requires many extra characters per entry.

Good thing is that XML is supported out of the box so converting legacy components to a new format should be very straightforward and users should not experience any problems. This can be done automatically in background:
- If JSON/YAML exists open it
- Else try to open XML
    - Data is loaded and used, on next save store it in new format
    - After storing in new format remove old file

Caellian, 2020-02-19:
I was stuck for a long while not being able to progress on this issue.

closed 2021-07-14 by BFH-ktt1