logisim-evolution/logisim-evolution #201: FPGA commander: Skip VHDL generation

BFH-ktt1, 2018-12-18:
The FPGA commander has the "Skip VHDL generation" checkbox. The function of this checkbox is to be able to download a generated bit-file directly to the board when it has been already generated (to avoid the time lost by the synthesis and place&route). 
In version 2.14.7 and 2.15.0 I observed that it is not working any more for Intel-FPGA based board (did not test it with xilinx boards).
As probably most people did not know it's functionality it is for me low priority, but it's functionality has been lost somewhere between 2.14.4 and 2.14.7.
to be fixed

maehne, 2018-12-19:
Well, I am aware of this function and showed it to my students, too. It is a real time safer when demonstrating circuits on our Gecko-Board during a class.

BFH-ktt1, 2018-12-24:
Fixed by my latest pull-request

closed 2018-12-24 by BFH-ktt1