logisim-evolution/logisim-evolution #219: macOS: CMD-K shortcut for "Simulate -> Ticks Enabled" menu action not working as expected

maehne, 2019-01-29:
On macOS, the menu action "Simulate -> Ticks Enabled" is bound to CMD-K. So, I would assume that pressing CMK-K would turn on the clock and pressing CMD-K a second time would turn the clock off again. However, currently, you have to keep CMD-K pressed for the clock keeping running, i.e., it behaves very similar to "Simulate -> Tick Once" (CMD-T) and "Simulate -> 'clk' Tick Once" (F2).

davidhutchens, 2019-01-29:
I noticed this a while ago. In fact, @maehne if you run the app you downloaded from my web site, you'll see that cmd-k works, but only for Ticks Enabled. I researched the issue and built a workaround. The actual problem is in the Java runtime. A previous bug report about double actions occurring on menu item command key selection was, it seems, only partially fixed. Regular menu items now cause a single action. But the fix was not applied to the Checkbox item. In Logisim-evolution, you'll notice that the command key equivalent for any Checkbox item fails. In debugging it, I found the double action using breakpoints in debugging. I then found  Java bug reports of similar things for other menu items marked resolved.

The reason it acts like a single tick is that it sends the action, causes a tick, then re-sends the action turning off the Checkbox.

What we should do is build a trivial app that demonstrates the Java bug and file a bug report with them.

My workaround was to replace the Checkbox with a regular menu item that changed its name. That isn't pretty and changes the look for Linux and Windows too. To fix it just for the Mac really creates ugly code by generating different object types for each platform.

davidhutchens, 2019-01-29:
We don't actually need to report this bug:
[https://bugs.openjdk.java.net/browse/JDK-8216971](https://bugs.openjdk.java.net/browse/JDK-8216971). It has already been reported.

davidhutchens, 2019-04-05:
This issue is fixed in openJDK 13 early access build 15, released 4/4/2019. Note this is a development build and not a general release.

I downloaded it to try. It works for me running the latest Logisim-evolution develop branch code built in Java 12, but run with Java 13. I then built Logism-evolution with Java 13 and that also works.

maehne, 2019-04-05:
@davidhutchens: Thanks for reporting that this issue will be fixed finally in OpenJDK 13!

davidhutchens, 2019-09-19:
OpenJDK 13 has now been released. The command key actions work correctly in Java 13. Should this issue be closed?

maehne, 2019-09-23:
Thanks @davidhutchens for testing whether the final version of OpenJDK 13 fixes this problem! I can confirm that the keyboard shortcuts now seem to work as expected. Therefore, I am closing this issue.

closed 2019-09-23 by maehne