logisim-evolution/logisim-evolution #222: Languages and documentation

BFH-ktt1, 2019-02-09:
One of the great features of logisim is the fact that it can be switched to different languages (Carl, thanks  for this!). Since the original version of Carl Burch, quite some effort was done on German, French, Spanish, and Italian was added. However, we still lack behind drastically in the other languages like Russian, Greek and Portuguese. 
Who is interested to help out by either updating an existing language or, if you want, add a new one?

mbaillif, 2019-02-09:
I have made a short review of the translation status of the Guide's documentation.
I will integrate it in the doc part of the project soon.

I have also developed a small program to help translate resources. He still lacks a bit of finishing. To be continued

Can be created a folder "translation kit" or we will find the state of the place and a minimum of documentation "how to do"

BFH-ktt1, 2019-10-20:
In my latest push I translated all points in the property files that were still empty using deepl. As I also generated completely the dutch translation I observed (as dutch native speaker) that there are still quite some improvements to do in the translations.

lohikarhu, 2021-05-16:
Hi there;  I noted some errors in the English documentation, and some 'Library' things completely missing, like POR, an important function (if implemented?)
I am able to commit some limited time to editing English-language issues, and I am a native English-speaker (not USA, not UK) with international work experience and some academic editing.. If I can help, it would be nice to be able to contribute in some small way.

lohikarhu, 2021-05-16:
Just going through the little tutorial for XOR simulation, I  noticed that the part about adding input and output pins has incorrect 'icons' for the current version... attached a tiny screenshot of that bit...
[image]
and the current icons
[image]

mbaillif, 2021-05-17:
Thank you very much.
I will correct this.

And all help is welcome.
The English version deserves proofreading and corrections.
In fact it is an endless work.

It is with great pleasure that I wish you a warm welcome.

lohikarhu, 2021-05-18:
Thanks... I'll try to do a bit at a time.

Nice project, and I was interested to see several YouTube videos using this
as a tool for several purposes.

It would be nice if there was a CMOS library, too, but libraries are a lot
of work!

Merçi

lohikarhu, 2021-05-18:
What could I do with the 'missing' functions? Are tehy described in another language, or are they still 'to be dtermined' ?

BFH-ktt1, 2021-05-18:
@lohikarhu , Basically the English version of the documentation is the most complete as this is the first one which is updated, the others are "following". Hence if it is not described in the English version than it is most probably not described at all.

maehne, 2021-05-19:
Thanks @lohikarhu for your offer to help improve the documentation! Feel free to ask any questions regarding the undocumented components here and we'll try to answer them.

mbaillif, 2021-05-19:
Hello,
I don't know if you need help getting started. Maybe you will save time with this information
- The doc is in HTML for compatibility with javahelp.
- It is in the path src/main/resources/doc

I see two approaches.
A) You know how to use Github and in this case you have to create a fork of reds-heig /
logisim-evolution branch develop
The documentation is in the path src/main/resources/doc
Make commits and pull requests to publish.

B) You just want to contribute and html and github are not your choice.
Go to this site: www.baillifard.com/logisim
On the line of the page click in the box of the desired language, the html page opens and with a copy paste place it in a word processor (libreoffice) or a text editor like Geany or Notepad++.
Edit, correct and send the document to the email address indicated at the top of the status page.

This procedure is still provisional and I don't despair of coming up with something better one day.

lohikarhu, 2021-05-19:
Great.... I'm not a GitHub user, in general, so, if it's ok, I can use method 2, but I've also done some 'bare metal' HTML.

I was poking around the POR block today, and it needs to be described, as its implementation is not as straightforward as say 'n seconds'.

My wife is in hospital at the moment, so my 'focus' is a bit 'other-where'.

Thanks for your quick and helpful replies!

lohikarhu, 2021-05-19:
Thanks, Torsten... I have to get a feel for the software, so I'll look at
some tutorials and such, but I can also do some 'proof-reading' of the
English doc's.

There seem to be some things that are not too 'intuitive' for a 'hardware
guy', but I'm kind of a 'system guy', so might be able to figure some
things out ;-) Have a fine day!

maehne, 2021-05-20:
Thanks @lohikarhu and @mbaillif for your willingness to work together on improving the documentation!

MarcinOrlowski, 2022-01-10:
I'd like to move this thread to discussions. Please object if I miss a reason it should stay as issue ticket.

maehne, 2022-01-23:
I have no problem with moving this to "dev talk".