logisim-evolution/logisim-evolution #229: Random Generator Oscillates

LordDecapo, 2019-02-22:
I have found that when you have the following conditions, a Random Generator will Oscillate. 
Reset - 1
Enable - X
Clock - X
[image]

[Dropbox link for a circ](https://www.dropbox.com/s/o5c2u7okmz1pfmf/RNG_Issue.circ?dl=0)

Used Logisim v2.15.0 on Windows 10.

BFH-ktt1, 2019-02-22:
@LordDecapo : This is for me a known issue and on the list to be fixed. For the time being the problem can be solved by putting for the attribute "seed" a number different than "0". Hope this helps for the time being, I will try to find a fix asap.

BFH-ktt1, 2019-02-22:
Fixed with my latest pull request.

closed 2019-02-22 by BFH-ktt1