logisim-evolution/logisim-evolution #231: Merging progress of the Holy Cross version into Evolution

BFH-ktt1, 2019-02-23:
I opened this issue to keep track of the changes Kevin Walsh has made on his Holy Cross version of the Evolution fork since Dec. 7th 2015. Bit by bit I am merging his features into the current development tree of evolution trying to keep backward compatibility between the two/three versions (classic,holy cross, and evolution).
Work done so far:
Push 30306e1: All, except for the changes to Terasic_DE0.xml, changes to BFHPraktika.java, addition of hex2sevenseg.java, hex2sevensegHDLGeneratorFactory.java, PortHDLGeneratorFactory, and PortIO.java.
Push 6fdb581: None
Push 4c6fec4: None (merged later with different appearances)
Push ba88937: None
Push 364e5f1: None
Push a02FF62: All
Push 651752d: All
Push ea851b3: All
Push cbd6612: Set for Holy cross edition when read in
Push a372887: Set for Holy cross edition when read in
Push ecb5c12: Skipped
Push 2ee7f20: Skipped
Push 8b5ffe1: All -> kept backward compatible with evolution by moving this changes to com.cburch.logisim.vhdl.* by duplicating some files
Push fa20b67: All
Push b9345bd: All
Push 2f02a00: Will not merge
Push 8ecc432: Will not merge
Push d9021f3: Not required
Push 47cfae0: Skipped
Push 63a1dcb: Merged
Push fc07b7a: All
Push f961e88: All
Push eb2dbdb: Will not merge

R3dst0ne, 2023-09-18:
I just noticed that @kevinawalsh recently contributed some commits to `Logisim-Evolution (Holy Cross Edition)`.
Among them are some interesting ones:
* auto-backup feature  https://github.com/kevinawalsh/logisim-evolution/commit/c57e32d893c0a6144c8974c021ad873c962ef43a
* Dynamic visibility for custom appearance shapes https://github.com/kevinawalsh/logisim-evolution/commit/cd8172936928f59f4196cefaa3c2e4ec0d97d8fe

maehne, 2023-09-28:
It would be nice to see progress in the backporting of @kevinawalsh improvements to his Logisim-evolution fork.