logisim-evolution/logisim-evolution #250: FPGA Commander: Netlisting gets stuck if components get connected to splitter inputs that are set to None

maehne, 2019-04-04:
I submitted a test case "Watch_PZ_12_mht1.circ" to @BFH-ktt1 where net listing in the FPGA commander gets stuck in a sub-circuit after I tried to fix "Found a sink without a source" warnings by connecting the open inputs to ground in that sub-circuit. At this point, it is impossible to abort the net listing even though the schematic entry of Logisim is still usable. One has to kill Logisim-evolution and start again to continue the work. I then noticed in the concerned sub-circuit that the splitter inputs, which I connected to ground, had been set to "None".

maehne, 2019-04-04:
I was able to reduce the test case (see attachment).
[netlisting_bug.zip](https://github.com/reds-heig/logisim-evolution/files/3042738/netlisting_bug.zip)

BFH-ktt1, 2020-03-21:
Fixed in my latest push, please re-open if it doesnt solve the problem

closed 2020-03-21 by BFH-ktt1