logisim-evolution/logisim-evolution #252: RAM doesn't have a clr pin

CoCoThink, 2019-04-10:
In old version of logisim2.7.1 , there is a clr pin to clear all data, but it is lost in the logisim-evolution, can it be added back?

BFH-ktt1, 2019-04-10:
Dear CoCoThink,
The clear pin was removed as real memories do not have this possibility (e.g. to clear a RAM each individual entry must be written to 0). Nevertheless I will reconsider when going over the memories to add it in as an option (meaning it can be enabled or disabled by an attribute). Please stay tuned.

PixelToast, 2019-05-17:
This is false, many SRAM chips have a way to clear the entire memory and most embedded NAND allows page-level clearing

For example http://www.pyramidsemiconductor.com/download/P4C150.pdf
You can also reset any SRAM chip quickly by simply cutting the power

BFH-ktt1, 2020-01-07:
Implemented in my latest push

closed 2020-01-07 by BFH-ktt1

sear9487github, 2022-11-03:
where is the clr pin? i cant find it anywhere.

davidhutchens, 2022-11-04:
@sear9487github: When "Use clear pin" is set to Yes, it appears near the left side of the top.