logisim-evolution/logisim-evolution #254: Allow it to zoom out more of Logisim schematics

maehne, 2019-04-12:
Sometimes Logisim schematics get very wide and won't fit on a screen if manually zooming out. The zoom is currently limited to be >= 20 %. The `Auto` button already allows to zoom out below this value.

MarcinOrlowski, 2021-07-14:
I reworked zoom controls so now you can go down as low as 5%. I also added 3 levels of granularity, so zoom step is smaller for lower values and increases for bigger zooms: step 5 up to 50%, then 10 up to 200% and 20 up to 1000%

R3dst0ne, 2021-07-26:
Since #814 zooming in/out with the thumb pad can lead to very precise numbers.
[image]
[image]
Using the `Auto` size button also sets very precise numbers.
[image]
In my opinion, the zoom values from these actions should align with the other types of zooming.

MarcinOrlowski, 2021-07-26:
Oh :( I thought I was rounding this so something sane. Apparently not. Will fix it.

PR #814 (2021-07-15): Reworked Zoom controls. Closes #254
Reworked Zoom controls for wider ranges and better granularity. Also added quick zoom buttons and change the way zoom value is shown.

[image]

Closes #254
commit 5afde2f7 (2021-07-15): Reworked Zoom controls. Closes #254
commit d7d56ee9 (2021-07-15): Merge pull request #814 from MarcinOrlowski/zoom-improved

Reworked Zoom controls. Closes #254
closed 2021-09-09 by MarcinOrlowski