logisim-evolution/logisim-evolution #255: Auto-pan while wiring or placing components

maehne, 2019-04-12:
It would be nice if Logisim would automatically pan into the direction of the mouse if the latter hits the screen border while one is wiring or placing components. Currently, one has to use the arrow keys for panning in this situation.

emrankb, 2019-12-10:
@maehne done!

emrankb, 2019-12-10:
@maehne Hello, Are you going to close the issue? :)

davidhutchens, 2019-12-10:
Thank you @emrankb for the fix. My testing shows that the fix works for dragging an existing object. However, unless I am doing something wrong, it did not work for placing new objects.

maehne, 2019-12-11:
@emrankb: Thanks a lot for this fix. It works well for moving already placed objects and drawing new wires. I can confirm @davidhutchens observation that it does not yet work for placing new objects. It seems like there is some special handling allowing to move the mouse in the library/properties panel. However, I think the more consistent behaviour would be to have the auto-pan behaviour also when placing new objects. Anyway, you can easily end placing new object using the `ESC` key.

Therefore, I propose to keep this issue open until this new feature works in all sensible cases.