logisim-evolution/logisim-evolution #259: Multiply and Divide circuit default should be unsigned

davidhutchens, 2019-05-01:
Recent changes to the develop branch allow the multiply and divide circuits to be either signed or unsigned. This is a nice addition. But the documentation and older versions of Logisim and Logisim-evolution only had unsigned. The current default is signed. This causes old circuits to default to signed when open and thus breaks old designs. I think it would be better if the default were unsigned so that existing files will open correctly.

BFH-ktt1, 2019-05-01:
@davidhutchens : You're completely right, I adapted it in my latest push (on the tikz exporter that will be merged shortly with develop), thanks for the input!

closed 2019-05-02 by BFH-ktt1

davidhutchens, 2019-05-03:
@BFH-ktt1 : Today's merge to develop changed the default for the multiply circuit to unsigned. However, the divide circuit still defaults to signed. It should also be unsigned.

BFH-ktt1, 2019-05-03:
@davidhutchens : Ooops missed the divider, should now be fixed.