logisim-evolution/logisim-evolution #264: Symbols for the controlled buffer and controlled inverter are wrong in IEC (rectangular)

BFH-ktt1, 2019-05-05:
The current symbols used when selecting IEC in the preferences stay as ANSI. As changing them to the correct ones would change their size, backwards compatibility would be broken. I leave this as open issue until I figured out a correct way to solve it.

maehne, 2021-08-25:
@BFH-ktt1 and @mbaillif: Maybe it would make sense to attack this issue in several steps:

1. Show in pictures the current situation in Logisim-evolution and highlight the issue with respect to the official IEC symbol.
2. Post proposals how the IEC version of the controlled buffer/inverter could look like in Logisim-evolution.
3. Discuss the proposals and how to implement while trying to avoid as much as possible/reasonable backward compatiblity.

Maybe, we should also start thinking about how to approach breaking changes in the long run. E.g., if we need to modify pin positions/size of some components, which would break loading old schematics using them, we could check for this while loading and then guide the user in updating his schematics by showing him one instance of the affected compent(s) after another, proposing an automatic resolution, and giving him a chance to fix/tune the local routing manually. What do you think?

maehne, 2026-10-01:
Thanks @hewzhew and sorry for my late feedback to your analysis of this old issue! Indeed, fixing it with the current constraints that a switch between ANSI and IEC shape does not have an influence on the pin position is difficult. The IEC tristate buffer symbols, which I found [[1](https://de.wikipedia.org/wiki/Tri-State), [2](https://international-university.eu/dispert_website/Digital_Design_Website_German/digital9/dig009_2.htm)] have the enable input above the data input. With the current constraints, we would need to deviate from that and leave the enable input below. Then we would need to add the small triangle pointing down before the output and therefore would need to shift the "1" to the left.

I think one alternative for gates, which currently cannot fully comply to the look of the IEC standard, would be to add a new selectable appearance, e.g., "Logisim-evolution strict IEC", "Logisim-evolution IEC" or simply "IEC". If the IEC shape is selected in the application preferences, newly placed gates would default to this new appearance and already placed gates would be left alone / would need manual change of their appearance by the user. Then, we would not need to invest the effort for automatic migration.

To estimate the effort, we should also first check whether other gates besides the controlled buffer have a non-IEC-conformant shape. Then we should define the improved shape within the constraints of the canvas grid, agree on these draft sketches, before implementing them.