logisim-evolution/logisim-evolution #271: Gated clock warning on banked registers

LordDecapo, 2019-05-27:
So I have run into an issue with I believe is the register component.  Attached is a file where if you go to the FPGA commander and go to download it using the settings in the screenshot... it will spit out warnings related to intermediate gated clocks and the sinks for those gated clocks. This only shows up as a warning when you compile this file alone, however, if you add it as a library to a larger project the warning turns into a fatal error. I can not for the life of me figure out what is causing this problem. As of now, I can not flash my FPGA with my larger project due to this issue. 

[image]

[GatedClockTest.zip](https://github.com/reds-heig/logisim-evolution/files/3225011/GatedClockTest.zip)

BFH-ktt1, 2019-05-28:
As long as the "clock" line is connected to a pin and not a "clock" component from the "wiring" library, logisim is interpreting the line as gated clock. If there are some instances that are connected to a "clock" component and others not logisim will fail and put out the fatal error as it is in this case not possible to generate a correct HDL-netlist with the current implementation of logisim. Hence the solution to your problem is to connect all those "clock" inputs to a clock component.
Hope this helps.
B.t.w. clicking on the DRC-symbol left in the warnings/error tab will highlight in your schematic where logisim found the error. This helps tracking down the problem.

LordDecapo, 2019-05-28:
Thank you so much!!! That explains everything and has cleared all my errors :D

I was trying to use a modified dual port reference VHDL component as a custom HDL entity, but the code wouldn't validate... I don't need it now (put the code below in case someone can see why it's not validating, online validation tools say its all good)
`library ieee;
use ieee.std_logic_1164.all;

entity true_dpram_sclk is
    port 
    (    
        data_a    : in std_logic_vector(7 downto 0);
        addr_a    : in natural range 0 to 63;
        addr_b    : in natural range 0 to 63;
        we_a    : in std_logic := '1';
        re_b    : in std_logic := '1';
        clk        : in std_logic;
        q_b        : out std_logic_vector(7 downto 0)
    );
    
end true_dpram_sclk;

architecture rtl of true_dpram_sclk is
    
    -- Build a 2-D array type for the RAM
    subtype word_t is std_logic_vector(7 downto 0);
    type memory_t is array(63 downto 0) of word_t;
    
    -- Declare the RAM
    shared variable ram : memory_t;

begin

    -- Port A
    process(clk)
    begin
        if(rising_edge(clk)) then 
            if(we_a = '1') then
                ram(addr_a) := data_a;
            end if;
        end if;
    end process;
    
    -- Port B
    process(clk)
    begin
        if(rising_edge(clk)) then
            if(re_b = '1') then
                q_b <= ram(addr_b);
            end if;
        end if;
    end process;
end rtl;`

closed 2019-05-28 by LordDecapo