logisim-evolution/logisim-evolution #277: Logisim ITA component Programmable Generator: I cannot enter zero value as initial data

maehne, 2019-08-30:
> When I try to send a low signal as the first signal of my generator, In the high signal column I place the value zero but it does not accept it and forces me to send high signals at the beginning of the transmission

This issue reported by @ArielG100 as [issue #9](https://github.com/LogisimIt/Logisim/issues/9) was reported to the [LogisimIt GitHub repository](https://github.com/LogisimIt/Logisim/) and also affects Logisim-evolution.

BFH-ktt1, 2019-12-29:
It is even worst, the programmable generator is broken in evolution

maehne, 2020-04-06:
This and some other Logisim ITA issues seem to have been fixed by its author in commit [c047de94](https://github.com/LogisimIt/Logisim/commit/c047de940b99a0fd8e89f9f178ace1980f8fd1a4). We should check if they apply to Logisim-evolution, too.

maehne, 2020-04-07:
Some additional fixes in Logisim ITA were done in commit [bf526841](https://github.com/LogisimIt/Logisim/commit/bf52684130cbbbe041d6cb7d12ac8ee92cf8e082).

maehne, 2020-05-12:
Some additional fixes in Logisim ITA were done in commit [e3500f86](https://github.com/LogisimIt/Logisim/commit/e3500f8641acfabb77b609f7fc40e418fe20b9cf).

closed 2021-07-14 by BFH-ktt1