logisim-evolution/logisim-evolution #278: Combinational analysis

teddfirth, 2019-09-14:
Hello,

I'm switching over to using Logisim-Evolution from the legacy Logisim app and I'm having difficulty accessing the combinational analysis feature.  From what I've been able to read on other posts on this issue, it seems that this feature was disabled intentionally but can be accessed using a command line instruction.

I'm running version 2.15 of Logisim Evolution and the application is labeled 'logisim-evolution.jar'.  I'm using a Macbook Pro and have limited experience using the Terminal.  My attempts to get the combinational analysis feature working have been unsuccessful so far.  Any suggestions or help would be greatly appreciated.

Thanks!

maehne, 2019-09-14:
To launch Logisim-evolution from the command line, you have to issue:
```
java -jar logisim-evolution.jar
```
Passing the `-help` option, will summarise the possible command line arguments:
```
java -jar logisim-evolution.jar -help
```
The combinational analysis feature gets activated with `-analyze`:
```
java -jar logisim-evolution.jar -analyze
```
You may consider rebuilding Logisim-evolution from the [develop](https://github.com/reds-heig/logisim-evolution/tree/develop) branch using the instruction on the project page. You will get a considerable number of improvements and fixes. Especially, on macOS, you can build an DMG image, which contains a convenient app bundle to launch Logisim-evolution. If you want the analyse feature to be activated permanently, you can have a look into the file `src/main/java/com/cburch/logisim/Main.java` and change the corresponding static member variable in line 97.

teddfirth, 2019-09-14:
Thanks for the quick response!

When I enter 
`java -jar logisim-evolution.jar -analyze`

in the command line, I receive the following:
`Error: Unable to access jarfile logisim-evolution.jar`

The logisim-evolution.jar file is located in the applications folder on my computer.  When I double click on its icon in the applications folder, it opens without issue.

And thanks for the suggestion regarding rebuilding logisim-evolution from the develop branch.  I'll start exploring that, although my skills with that sort of thing are pretty basic.

Thanks again for the response and there are any thoughts on why I'm unable to access the jarfile, it'd be much appreciated.

maehne, 2019-09-14:
I haven't seen the above error message yet. However, it may be that `logisim-evolution.jar` is simply not in your current working directory. In that case, you have to make sure that you provide the full path to `logisim-evolution.jar` when you pass it as argument to the Java interpreter. Another thing you may check is that the access rights to the file are set to `rw-r--r--`. You can check this with `ls -al <path-to-directory>`.

teddfirth, 2019-09-14:
Thanks again for the quick reply and the suggestions—I'll give them a shot and see if I can get it to work.

teddfirth, 2019-09-20:
Thanks, I was able to set the correct path to `logisim-evolution.jar ` and got it working as needed. Thanks again for the help!

closed 2019-09-20 by teddfirth