logisim-evolution/logisim-evolution #284: Windows 10, combinatoric analysis: inversion bars not correctly aligned over variables

maehne, 2019-10-11:
In the Expression and Minimization tabs of the Combinational Analysis window, boolean equations are not correctly displayed when `Mathematical` is selected as notation. The inversion bars are not correctly ligned above the respective variables as can be seen in this screenshot:

[image]

I am running the latest Logisim-evolution version 3.1.0 built from the `develop` branch (commit 189f771) on Windows 10 with OpenJDK 13.

simoc, 2021-03-15:
I think this bug is in method `ExpressionRenderData.style`.

The font `Monospaced` is used for the display of the expression in the Expression tab. Although ASCII characters in this font are fixed width, the Unicode dot symbol `\\u22C5` is not fixed width.

Before the width of the text on the screen is calculated in this method, all dots are replaced with underscores, and this will give an inaccurate width.

maehne, 2021-03-19:
Thanks @simoc for fixing this issue with your PR #525! This issue can be closed.

PR #525 (2021-03-17): Fixed alignment of inversion bars in Combinatorial Analysis Expressions
Removed special logic for converting parentheses and dot characters to underscores before calculating width of expression text.

Java returns the correct widths for strings containing these characters, so no special logic is needed.

The only special case is spaces. Java will return the same width for "abc" and "abc ", so we do need to convert each space to an underscore, before calculating widths.

Attaching screenshots of expressions in #284 after my changes on Windows 10 and Ubuntu Linux, both with OpenJDK 11. I have no [... cut]
closed 2021-04-03 by BFH-ktt1