logisim-evolution/logisim-evolution #290: Add autocompletion for VHDL

Caellian, 2019-11-30:
This is a minor enhancement, but adding autocompletion and shorthand notation for vhdl would make using RSyntaxTextArea more comforable. There's a very simple to use [library](https://github.com/bobbylight/AutoComplete) compatible with it.

maehne, 2019-12-01:
This would be a nice feature. Though, we should ask ourselves how much effort we should invest into improving the code editor(s) (Besides VHDL and Verilog, there would be also a need to support different Assembler languages) in Logisim-evolution. Most people have their preferred text editor and tend to copy and paste from it into the (usually mediocre) tool-specific Editors. Still, I reckon the value of an integrated tool especially for beginners, which is the target audience of Logisim-evolution.

maehne, 2026-09-24:
@hewzhew: Thanks for offering to implement this feature. The approach you outline sounds mostly good. When I looked up `com.fifesoft:autocomplete`, I couldn't find any support for VHDL, Verilog, or the Assembly dialects offered by Logisim-evolution. So, the effort might be higher than just integration this library. If enhancement to the language support are needed, they should be ideally upstreamed. 

`com.fifesoft:autocomplete` version 4.0.0 requires JDK 25. IMHO, it is fine if Logisim-evolution would in consequence also rely on JDK 25, as it is an LTS release.

In any case, my initial comment above still holds. I think the value of the enhancement is limited, as people should use a proper external editor for serious coding.