logisim-evolution/logisim-evolution #291: Remove RSTA jar and use maven

Caellian, 2019-12-01:
RSyntaxTextArea dependency is currently consumed as a jar and is available through maven.

Differences between the jar and upstream versions are:
- Language support for languages which aren't used by this project are removed.
- VHDL support is added using JFlex (repo) generated Marker class

## What is required to do this?
RSTA needs to remove integrated support for languages and move each language into it's own repository and package. (See [#322](https://github.com/bobbylight/RSyntaxTextArea/issues/322) on RSTA repo)

## Further details
RSTA doesn't support new versions of JFlex so supporting JFlex marker class generation through Gradle is most likely a wasted effort at the moment.
Further, JFlex project has no Gradle support which means the build file would have to import it as a buildscript dependency (~8 lines of code) and define a custom task for generating the source and cleaning it up (~20 lines of code + semi-complicated regex).

### JFlex vs BNF
What I think RSTA needs is a [BNF](https://en.wikipedia.org/wiki/Backus%E2%80%93Naur_form) support. Most basic BNF would be good enough for VHDL. Implementations of BNF greatly vary and though [EBNF](https://en.wikipedia.org/wiki/Extended_Backus%E2%80%93Naur_form) is standardized, it's far from ideal.
W3C uses [it's own](http://www.w3.org/TR/REC-xml/#sec-notation) for XML. This problem is out of scope of this issue and has more to do with the fact that there's no proper standard for defining language syntax.

BNF would be a much better solution as it's simpler to maintain, has more languages already defined for it and is overall a lot simpler format than JFlex input files.

Either way, a partially automated way of defining languages in unarguably a bad design choice.

## Temporary solution
- Using a highlighter generated by JFlex, manually patched and fixed to cooperate with RSTA
- Providing JFlex source file in case RSTA starts supporting new JFlex versions and/or VHDL highligher needs to be recreated.

I believe Gradle can remove files from dependencies, as RSTA is accessing it's highlighters using reflection there shouldn't be any errors as long as predefined syntaxes aren't used (they currently aren't). I'll fiddle with this a bit some time in future and see how it goes.

maehne, 2019-12-01:
@kevinawalsh: Maybe, you could comment on this issue, as you implemented originally this feature in your own Logisim fork?

kevinawalsh, 2019-12-01:
I'm not entirely sure what the bug/issue being reported here is. Perhaps @Caellian could clarify?
In the meantime, some comments:

> Language support for languages which aren't used by this project are removed.
The stripped RSTA jar being used by logisim is just over 512KB (compared to 38MB for the latest windows EXE distribution of logisim). Every byte counts, but this seems like a low priority if there is much effort involved.
> VHDL support is added using JFlex (repo) generated Marker class
I don't know what a "JFLex (repo) generated Marker class" is. Is this RSTA specific? Regardless, the current logisim VHDL keyword hilighting (minimal as it is) seems sufficient for most purposes.

As for the rest of the issue (starting at "What is required to do this?" and continuing from there), I think I just don't understand what is being proposed here. I don't use maven or gradle at all. I don't use a new version of JFlex either -- an old version was used once to generate the syntax rules for VHDL. Perhaps logisim should just plan on not ever updating the RSTA dependency? I had no plans to update in the foreseeable future anyway.

Caellian, 2019-12-02:
Issue is that RSTA is being included in the repository as a jar file instead of being a maven dependency. It adds 1.1MB when the repository is being pulled when you could effectively include same information in several bytes by using "implementation(...)" statement in build script.

Here's a list of reasons why dependencies should not be included in the repository as jars:
- Using maven allows users to download ONLY your code and rest of the code is downloaded only if they wish to build it. This poses a big problem for large projects consuming a whole range of dependencies some of which also have their own transitive dependencies. It clutters your project, bloats project size, forces users to download dependencies they might not wish to download.
- Updating a dependency is very hard and complicated when you're using a local hack you compile and bundle with your code.
- If developers are using RSyntaxTextArea in some of their other projects, including it with maven allows Gradle (and most other sensible build systems) to cache that dependency instead of downloading it with each repository individually.
- If you updated RSTA in it's current state, everyone who pulled your latest commit would have to download the entire jar again (as git can't diff binary files). Alternatively, you can keep the code related to that dependency in your project in which case git can generate diffs and only download the actual changes you've made.
- From perspective of a contributor, If I (in this scenario) wanted to contribute to the VHDL code editor, I have very little context to work with when it comes to how dependencies work. Sure, I took the effort to search for what was changed and in this case it wasn't much. Others might not be willing to spend half an hour decompiling dependency jars to see what rsyntaxtextarea.jar contains and what is different. This jar could be version 1.0.0 or you own modification of 4.0.0 where you removed all existing code and replaced it with methods that have very specific behaviour. IntelliJ IDEA has fernflower integrated so I can go through jar code and see how it works, but official documentation no longer fully applies. On other hand, when I see 'com.fifesoft:rsyntaxtextarea:4.0.3' in the build script, I know RSTA by com.fifesoft, version 4.0.3 is used. I can go to GitHub and see source code there, maybe even contribute to RSTA.

RSTA may introduce new features, optimizations, bug fixes down the line. Users like programs that work, not programs that are overrun with bugs that have been fixed in newer library versions (even minor versions might introduce improvements).

Also, I just checked, RSTA jar in lib directory is only 14KB smaller than jar that would be downloaded from maven (v3.0.4). Those additional 14KB wouldn't affect RAM consumption as JVM wouldn't load unused classes.

By "JFlex (repo) generated Marker class", I meant the class generated by JFlex from .flex file. "(repo)" part was supposed to be a link to project repository, I'll fix it in a second.

I admit RSTA isn't really the best example as it updates very slowly, but ideally lib directory should be empty. Sorry for the wall of text, hope this clarifies my reasoning behind this issue.

maehne, 2019-12-02:
@Caellian: Thanks for your clarifications! I agree with you that we should try to avoid distributing patched JARs and strive for replacing JARs in the `lib/` directory by equivalent dependency lines in the gradle build files. I clearly see the benefits regarding handling/updating the dependencies. We appreciate a lot your support in this respect for our project! Gradle works well and the build scripts are more readable and compact than before.

Regarding, the missing JFlex support for Gradle, did you check out the two options listed in their [wiki](https://github.com/jflex-de/jflex/wiki/Build-tool-integration)?

Caellian, 2019-12-02:
@maehne Didn't see the wiki. [Plugin by Jörg Prante](https://github.com/jprante/gradle-plugin-jflex) seems more up-to-date. It's written in Groovy but I think (not 100% sure, input on this is welcome) there's no performance penalty as it's compiled Groovy.
Either way, problem with having to modify generated sources (as described [here](https://github.com/bobbylight/RSyntaxTextArea/wiki/Adding-Syntax-Highlighting-for-a-new-Language#JFlexBased)) is still there. Regex patching would work but is not reliable and will likely break.
Another alternative I though of while writing the previous comment was to write a class that extends `AbstractJFlexTokenMaker` and make JFlex generated class extend that one instead of `AbstractJFlexTokenMaker` directly. I'm not familiar with how JFlex works either so it's best that this is done by someone who has experience or some knowledge using it.

maehne, 2019-12-03:
Looking at the [source code for syntax highlighting](https://github.com/bobbylight/RSyntaxTextArea/tree/master/RSyntaxTextArea/src/main/java/org/fife/ui/rsyntaxtextarea/modes) in [`RSyntaxTextArea`](https://github.com/bobbylight/RSyntaxTextArea),
I don't think that it is a good idea to invest effort into developing a BNF library for this single purpose. All syntax highlighters are there implemented using `JFlex`. The [developer documentation of `RSyntaxTextArea`](https://github.com/bobbylight/RSyntaxTextArea/wiki) seems to explain the process of adding support for new languages quite nicely. The creationg of the `JFlex` files is even supported by a dedicated tool called [`TokenMakerMaker`](https://github.com/bobbylight/RSyntaxTextArea/wiki/Adding-Syntax-Highlighting-for-a-new-Language#using-tokenmakermaker-to-generate-a-tokenmaker). Also from the [documentation](https://github.com/bobbylight/RSyntaxTextArea/wiki/Adding-Syntax-Highlighting-for-a-new-Language#plugging-your-tokenmaker-into-an-application), it doesn't seem to be necessary to patch the `RSyntaxTextArea` JAR at all, as a custom `TokenMaker` for a new language can be registered with the `TokenMakerFactory` that is to be used for a `RSyntaxTextArea` instance.

Therefore, I think effort is better spend in cleaning up the Logisim code to follow upstream's recommendation on how to customize `RSyntaxTextArea` for an application's needs. `RSyntaxTextArea` seems to be actively maintained and its developers seem to be open for pull requests. So, it could be worthwhile to upstream support for VHDL, Verilog, Nios 2 Assembler, and Risc V Assembler.

@Caellian: What do you think? Would you be willing to work on this issue?

BFH-ktt1, 2019-12-03:
Hi all,
Before starting work on this issue some background information from me:
1) The RSyntaxTextArea used in the SOC-library uses an automatically generated TokenMaker that depends on the CPU used, hence here there is no JFlex is involved. Moreover, the results of the TokenMaker are used in the Assembler to do syntax checking and assembly. Hence here we should in line with the BNF library, unless the TokenMaker class ABI changes.
2) For the VHDL-component, as stated earlier, I merged Kevin Walsh's work. Here, for the moment, there is the JFlex present, however:
a) At the moment I was merging this nice addition of Kevin Walsh I did not take time to look into the details, and just added the required dependencies. Starting the SOC library I got inspired by this addition and found that Kevin used RSyntaxTextArea for highlighting, which I than investigated and got to know (reminds me that I have to put an acknowledgement to RTA in the readme.md).
b) As, due to time limitations, I did not get into the JFlex file of Kevin, it is still in place. There are in my opinion two ways to go here, (1) the one proposed by @maehne to move the VHDL/VERILOG support to RTA as plugin, or (2) the solution described below:

Currently logisim has (all) VHDL and Verilog keyword present, as they are used for label checking (as labels should not be VHDL or Verilog keywords [this would result in syntax errors during synthesis and P&R]). Using these same lists to generate run-time a TokenMaker would remove duplicated code and makes the first "simple" high-lighter. Things like comments, strings, hex, etc. are relative easy to implement in the Tokenmaker (the one used in the SOC-library could be used as base). The advantage here is that it would be evident to add #290 and that it should be quite easy to extend the VHDL-component to be extended to Verilog (first step towards #274).

In case of solution (1) it would be nice to be able to read-out a list of VHDL/Verilog keyword to be used in the label name checking. In case of solution (2) I forsee one problem, being the fact that VHDL is case-insensitive, I did not see a possibility in RTA or Tokenmaker to specify case insensitivity (maybe I did not look deep enough).

Both solutions involve looking in the current JFlex to see what rules are implemented to port them.

These are my 3c, what do you think?

Caellian, 2019-12-03:
@maehne I wanted to create BNF library as it would also be useful later for another project I'm working on (which requires Java, Kotlin & Scala highlighting and JavaFX component for editing code). If it's not required to remove other languages I agree patching isn't required although it would be nice if RSTA was modular like [Jackson](https://search.maven.org/search?q=g:com.fasterxml.jackson.dataformat).

Also, I'd like to note that I'm currently very occupied with student duties and not at all familiar with VHDL or Verilog so someone else would be a better choice for implementing this - they just have to delete the jar file, uncomment RSTA implementation in build script and the only error that pops out shows the place where RSTA language needs to be selected at GUI initialization.

@BFH-ktt1 While solution (2) would be simpler and quicker to implement than learning JFlex, I believe JFlex generated classes might perform better.

Performance might be completely arbitrary. I'm not sure how RSTA handles highlighting - it only needs to process lines currently displayed on screen and if it's parsing entire source file, performance might be important as changing TextArea content requires re-parsing.

BFH-ktt1, 2020-01-29:
Although we now use the maven import (thanks @davidhutchens), there are still very important and interesting points left in this discussion. So I will leave it open.

closed 2021-07-14 by BFH-ktt1