logisim-evolution/logisim-evolution #303: Documentation

BFH-ktt1, 2019-12-17:
Hmmm, I also have to blame myself, but the current help system is out of date and needs some handling. Maybe a first step could be to first attack on the English version and than go to the other  languages. Many new features are not/partly documented and make the use of logisim a lot easier to the targeted public, when documented. It would be great if we could maintain Carl Burch's standard of documentation; who wants to help?

rpedroni, 2020-01-03:
Hi @BFH-ktt1 ! I'm a professor from Brazil and have been using the original Logisim for quite a while now in my classes but feel its lack of evolution stagnated what I can do with it. I'd love to help with this documentation, but I would like to get to understand better logisim-evolution more, since I've never used it before. If you can give me any starting point, that'd be great and hopefully I can help with this surely needed documentation.
Thanks!

maehne, 2020-01-04:
@rpedroni: Thanks for your offer to help! Contributions to the documentation are highly appreciated! Another user, @mbaillif, has started to work on updating the documentation for Logisim-evolution 3.2. You can see his contribution in PR #313. Once you know better Logisim-evolution and have ideas on which areas you would like to work, it would be good if you could let us and @mbaillif know, so that we can coordinate a bit the effort.

To get started with Logisim-evolution, I suggest that you build it from source from the [develop branch](https://github.com/reds-heig/logisim-evolution/tree/develop). I suggest that you build it against [OpenJDK 13](https://adoptopenjdk.net/?variant=openjdk13&jvmVariant=hotspot). Besides some new component models, which you can find in the various libraries, the most notable enhancements are the FPGA Commander and Board Editor enabling downloading on FPGA boards based on Xilinx (using the Xilinx ISE and Vivado toolchains) and Altera/Intel FPGAs (using intel Quartus Prime). The Combinational Analysis capabilities have been improved considerably as well as the capability to add new blocks described in VHDL. These are co-simulated in ModelSim or QuestaSim. These latter two enhancements were developed by @kevinawalsh in his [Logisim-Evolution (Holy Cross Edition)](https://github.com/kevinawalsh/logisim-evolution) fork, which have been back-ported by @BFH-ktt1 into Logisim-evolution. Other features have been integrated from [Logisim-ITA](https://github.com/LogisimIt/Logisim). Very recently was added the System on Chip components library allowing to do virtual prototyping for Risc-V- and Nios-2s-based SoPCs.

I hope these hints get you started to explore the enhancements of Logisim-evolution! Feel free to ask any questions arising while trying to use them.

maehne, 2020-01-04:
Another, currently stalled activity (due to lack of time) is the integration of the FSM editor developed by @sderrien. You can find the current state in PR #217. It needs to be rebased on the current develop HEAD and conflicts and build issues (due to our switch from Ant to Gradle) need to get resolved.

mbaillif, 2020-01-05:
Hello everyone.

On this temporary link www.baillifard.com/logisim you can observe the state of progress of the documentation.
I have not yet approached the library part.

In general, everything should be checked, especially the boxes with V and M 
There's still 1,000 and 1 mistake left. Obviously the red boxes have to be translated.

I'm gonna keep moving forward a little bit every day.

Text in html with css (a compromise to work with java.help v2)

I understand that it's not necessarily accessible to everyone. 
In this case please just send me the text (in Ascii or Free Office or Word.). I will format it.

Also if you have parts of new property that are not documented you can send me a short documentation so that I can develop on it.
For example everything about FPGAs.

@rpedroni: You can already browse through the above documentation in the column PT.
Each box is a hyperlink. The other chapters not yet adapted are accessible from the documentation pages. 

I would appreciate a review of the documents marked with an M, and feedback. 

Thank you.

rpedroni, 2020-01-07:
Hey guys, thanks for the response! Even though I'm an EE and teach EE, my main focus (for reasons to long to list here) has been web and mobile development. That being said, I was imagining we could use some of the documentation tools/frameworks already ou there (I really like [Docusaurus](docusaurus.io)).
If I tried following a path to fit the documentation in some framework like this, does that seem good? Are do we have to have some more loose file types for some readon (classes, etc.)?
@mbaillif I started checking out your docs, congrats for all the work put in to it :)
I'll be using these to check any subject I'm not very familiar with. This week I'm short on time but I'll get some hours into reading these and update you guys with any news.
Thanks!

rpedroni, 2020-01-07:
@mbaillif by the way, I translated the "About the program Logisim-evolution" to PT. Do you have an e-mail I can send you this specifically? Thanks!

TheEvilSkeleton, 2020-11-14:
@BFH-ktt1 I'd like to help, but the problem is that just launching logisim-evolution is already a terrible experience. Not everyone wants to run a program using the terminal.

I've created a flatpak for Linux (see https://github.com/reds-heig/logisim-evolution/issues/448). In short, Flatpak is a universal package manager for Linux. The flatpak I made comes with a `.desktop` (executable icons from the application menu), so we don't have to run the terminal all the time. This flatpak however is not in the Flathub repository, so it is not that easy to access it. In the issue, I mentioned that if you guys don't want to maintain it, then there's not much I can do.

That is just for Linux, though. Windows and macOS users will still have a much harder time to get it to work (source: my schoolmates). It would be awesome if you can provide .exe and .dmg that automatically install OpenJDK and come with its own executable icon.

Sorry for being a bit off-topic. A documentation is not that important when many people have trouble launching it to begin with.

maehne, 2026-08-21:
To not loose track of @mbaillif's work on the documentation, which he [advertised](https://github.com/logisim-evolution/logisim-evolution/issues/1479#issuecomment-1644294093) in issue #1479:

> Hello
>
> I need help to proofread the new documentation of the Memory library pages (10).
> 
> They are available on this link https://www.baillifard.com/logisim/
> and also on https://github.com/mbaillif/Logisim-Evolution-documentation
> 
> I've made a lot of changes to the layout and some text. It's really a lot of work.
> 
> The Library libs status link gives direct access to the current status of the documentation.
> If you click in one of the boxes, the pages can be accessed directly.
> 
> If you're really motivated, you can start proofreading all the pages with a _V.
> 
> This is a first attempt. There's still a lot to do.
> 
> If you're not too comfortable with html or Github, you can send me a Word or OpenOffice document (copy and paste from the web browser) with the corrections clearly visible.

I think it would be great if his effort could be merged back into our documentation. Unfortunately, we lacked time back then to help him review the documentation. Volunteers are welcome!

maehne, 2026-09-28:
Thanks @hewzhew for your review of @mbaillif's documentation work! I agree with your approach. Indeed, @mbaillif should give his explicit permission that his updates on the documentation performed in his repository can be carried over into our repository under GPL-3.0 -- ideally by adding the suggested LICENSE/COPYING file.