logisim-evolution/logisim-evolution #306: Avoid duplicate names on duplicate

rkoe, 2019-12-19:
When duplicating, for instance, a LED, the copied element has the same name as the source. This must be avoided, eventually by adding a suffix to the name.

This was solved for "copying" (see #55), but  not for "duplicating".

BFH-ktt1, 2019-12-19:
Dear @rkoe , which version of logisim are you using?
In the current development version 3.2.0 this issue should be fixed.

maehne, 2019-12-19:
I cannot reproduce your issue with Logisim-evolution 3.2.0 built from the develop branch. With that version, the label gets cleared on duplicated or pasted objects.

rkoe, 2019-12-19:
I've tested the lastest release (2.15.0).

rkoe, 2019-12-19:
Ok, I can confirm that this seems to be fixed in 3.2.0.
(And also, 3.2.0 now does not clear the labels if the element is copied to an other logisim-file, which 2.15.0 unfortunately did.)

It would be nice, though, if there would be an already-compiled 3.x version to download.

closed 2019-12-19 by rkoe

BFH-ktt1, 2019-12-19:
I will very shortly put a release 3.0,3.1, and 3.2