logisim-evolution/logisim-evolution #307: RAM/ROM "edit contents" broken

rkoe, 2019-12-19:
In 3.2.0, editing RAM/ROM is broken.
When I click on "Contents (click to edit)" or right-click on the RAM/ROM and select "Edit Contents...",
nothing happens.
This worked in 2.14.6 and 2.15.

maehne, 2019-12-19:
Thanks for the report! I can confirm this problem.

BFH-ktt1, 2019-12-19:
I tracked down the issue and figured out that due to the fact that a stopped at a certain moment with merging Kevin Walsh's additions to logisim, the hex-editor got in broken state due to changes to the menu structure (of course the holy-cross version does not have this issue, evolution has it as I did not merge all changes made by Kevin). I will look into it as soon as I have the bandwidth, but unfortunately till mid February 2020 I am very busy with my teaching activities. If you need this feature please continue using the 2.15 version.

BFH-ktt1, 2020-01-07:
Should be fixed in my latest push, if still problems persist, please reopen the issue.

closed 2020-01-07 by BFH-ktt1