logisim-evolution/logisim-evolution #308: Bug in version 3.2.0

mbaillif, 2019-12-20:
If in place a free text on a schema, even just one.
There's this message when you're savoring the circuit.
"Text component not found."

BFH-ktt1, 2019-12-20:
I can reproduce, thanks for the catch!

BFH-ktt1, 2019-12-29:
Fixed in my latest push, closing the issue, please reopen if the problem is not solved.

closed 2019-12-29 by BFH-ktt1

micahwagner, 2021-04-21:
every time I put text down in a schematic scheme and try to save, it says "Text component not found." for every text component in the scheme. This does not happen when I put text in a circuit appearance.

micahwagner, 2021-04-21:
I have the newest version of logisim-evolution

micahwagner, 2021-04-23:
the version I'm talking about is 3.4.4

micahwagner, 2021-04-23:
sorry for mentioning it again, I didn't know it was tracked already