logisim-evolution/logisim-evolution #309: rotate/mirror for all elements?

rkoe, 2019-12-23:
Some circuits can be drawn much cleaner, when some elements are rotated/mirrored (e.g. a mirrored adder, feeding its result back to a register).

The gates can be rotated, but unfortunately, many other elements cannot. So, I think it would be useful if more elements could be mirrored (e.g. arithmetic) and/or rotated.

(As workaround, I can of course create a mirrored/rotated element by creating a subcircuit, adding the element and creating a mirrored/rotated appearance.)

MarcinOrlowski, 2021-07-13:
I am closing this. The rationale is: we will have flip/mirror for single elements in foreseeable future, but, while rotating whole circuit would be great thing to have, it is pretty complicated task (internally) and not going to happen really soon.

R3dst0ne, 2021-07-13:
@MarcinOrlowski 
IMHO this should stay open since:
>we will have flip/mirror for single elements in foreseeable future

is no reason to close an issue. Until these things get included, it should stay open.

> while rotating whole circuit would be great thing to have, it is pretty complicated task (internally) and not going to happen really soon.

@rkoe didn't ask for this feature. Using it as an argument to close this issue doesn't make much sense.

MarcinOrlowski, 2021-07-14:
We got 118 opened tickets right now, some date even 2015. There's no point to keep most of them hanging opened forever if no effort have been made address them for years to. Reducing ticket backlog also lets keep the focus on tasks that are more important (not to mention it makes managing tickets list easier). While OP's original request makes sense as such, due to the LsE internals adding this feature is rather not trivial and won't happen in foreseeable future. It's also not high demanding feature either so closing this ticket makes no harm too. If you want to not forget about this one, we can consider adding TODO file with references to "closed tickets that make sense" but I am skeptical if that would make any difference nor increase implementation chance for listed tickets. It will be rather like cleaning the room by putting the trash under the bed :)

BFH-ktt1, 2021-07-14:
I agree that we should clean up open tickets, however, this one is for me not a ticket that should be closed. This is on my todo list for a while now: making all components rotatable. It is relatively easy to do, as it is just a question of the paint function and the location of the pins. I'll reopen it.
The reason that it was not addressed is just the limited time we have to do so. I'll go through the tickets in the near future and close those that either have been already resolved, or make no sense anymore.

maehne, 2021-07-14:
Once we get to the roadmap discussion, we prioritise our list of new functionalities, which we would like to add to Logisim-evolution in the near-term to mid-term future. At that point, we can check whether closed issues contain useful proposal and reference them. We can then also start assigning issues to milestones.