logisim-evolution/logisim-evolution #312: RAM/ROM components in logisim

BFH-ktt1, 2020-01-02:
Dear all,
I am looking into especially #307 to fix the current broken RAM/ROM components, however I face some choices to be made and need your help.
The first issue is that in evolution the RAM components use "byte-enables", whilst holycross went with "line-enables" (single dual quad). To be able to be backward compatible to both versions I can:
a) Use the holycross behavior for the clasic and holycross appearance, and the evolution behavior for the evolution appearance (my current preference).
b) Add an extra attribute which selects the behavior (evolution or holycross).

Furthermore, there are several issues posted, namely:
#307: This will be implemented by an attribute where the CLR-pin can be enabled or disabled (for both evolution and holycross behavior).
#119: Here my idea is to have an attribute that allows for an asynchronous read with a synchronous write hence the read-port will have the same behavior as an asynchronous RAM, whilst the write port only writes on rising/falling edge. Is this the required behavior?
#79: Here the big question is which modes are interesting:
a) Semi dual ported (one write data, one read data, with separate address for the write and the read port)
b) Real dual ported (all signals double)
c) Triple ported/quad ported, ....
#33: The addressing mode is not completely clear for me, can it be specified in a bit more detail?

Please give me some feedback on the above issues such that I can implement the new RAM/ROM components to the requirements of the users?

Thanks in advance and wish you all a happy 2020.

davidhutchens, 2020-01-04:
I believe the first point under Furthermore should reference #252.

maehne, 2020-01-04:
@kevinawalsh: Maybe, you would like to comment on this issue?

BFH-ktt1, 2020-01-04:
@davidhutchens : Thanks, updated the reference

BFH-ktt1, 2020-01-07:
Many issues solved, but still #79 and #33 are open, so I keep this issue open.

davidhutchens, 2020-01-08:
Nice work, @BFH-ktt1. While I was playing with the updated RAM, I noticed a couple of minor things.

The clear pin does not have a label. It took some looking around to find it. There isn't much room in the image for a label, so perhaps you thought about a label but were not sure where to put it.

The tool tip on the clear pin doesn't match the comments on reset for Registers regarding asynchronous behavior, but the clear input clears the RAM without a rising clock. Should it say asynchronous?

BFH-ktt1, 2020-01-08:
@davidhutchens : Thanks!
About your points, I fully agree!
Some of my reasoning:
The clear pin is in general (FPGA-memories, ASIC-memories) not present and has to be implemented by a FSM that clears one position after the other; however, as the community liked this feature I reintroduced it "quickly" (without thinking of a nice way to do it).
I am open for suggestions how everyone would like to have this pin shown up in classic view and in holycross/evolution view.
I am also open for Asynchronous/synchronous behavior of the pin, please let me know what/which behavior(s) you require.

closed 2021-07-14 by BFH-ktt1