logisim-evolution/logisim-evolution #316: Sequential circuits built from gates

NulAsh, 2020-01-07:
Will it ever be possible to simulate such circuits?
[image]

mbaillif, 2020-01-08:
[image]

Here is a solution for the first circuit

h1s0k429a, 2020-02-19:
Im trying to create D / T Flip Flops but the software keeps crashing on me or gives me error.
When I try to create the flip flips in online gate simulator tools, it works. please fix this ;(

mbaillif, 2020-02-22:
This time Loigisim is rather right.
In the help chapter "Value propagation" and "Shortcomings" the limits of logisim in this field are presented.

This circuit is particularly unstable, even in the real world. 
The one I built to see, oscillated at 16mhz with 74 series gates.

If you use this circuit it will work.
It's the same schematic, designed a little differently.

[image]

closed 2021-07-14 by BFH-ktt1