logisim-evolution/logisim-evolution #320: Timetable windows

mbaillif, 2020-01-09:
Studying the chronogram part of the 3.2 develop version.
for the documentation 
I notice that the "Timetable windows" part is no longer available.
My questions : 
- Has it been moved?
- will it come back?
- An alternative?

BFH-ktt1, 2020-01-09:
Dear @mbaillif,
Please do not do documentation on the chronogram part in evolution 3.2 yet, as:
1) Kevin Walsh did a very nice job on improving the chronogram in his holy cross version
2) This will be the next part that I will attack in my merging effort of synchronizing evolution with holy cross.
If you want to continue on the documentation of the chronogram please checkout the latest version of the holycross branch of logisim. In this case both holycross and evolution can profit from your great documentation effort (I will try to get the holycross chronogram into evolution as soon as possible (read dependent on the amount of my free time, which is in the coming month quite limited).

BFH-ktt1, 2020-01-09:
Dear @mbaillif ,
Also important, could you please send me your credentials such that I can add you as contributor to the documentation effort?
Thanks in advance!

closed 2021-07-14 by BFH-ktt1