logisim-evolution/logisim-evolution #321: Evolution still uses jar libraries that might be outdated and are available in maven

BFH-ktt1, 2020-01-09:
I had a small moment to look into the lib directory of evolution, and noticed that evolution uses libraries made by other persons. The issues that I see here is:
1) Unfortunately we do not mention the persons/projects we depend upon in evolution.
2) We might profit from developments in those projects.
3) Due to switching to gradle we might include those libraries by use of maven
Unfortunately I have very little (read no) experience in gradle/maven (I'm a hardware guy :-)). I would appreciate help on:
a) Sorting things out, such that we can use with maven the required libs and also credit the guys creating/maintaining those libs.
b) Looking into the effort to upgrade those libs to the latest versions and see what needs to be done in logisim-evolution.
If you fill up to this task, pleas do not hesitate to place a comment to this issue

davidhutchens, 2020-01-23:
I was planning to have a look at the deprecated Java features that currently generates warnings on compiling. To do that, I tried to import logisim-evolution into a recent version of Eclipse. That failed. After some research I determined I needed to update Gradle to version 6.0.1. I did that and succeeded in importing it. So it looks like an update to Gradle is also needed.

I too have little experience with gradle/maven, but I am a software guy so maybe I should learn more about it. If no one with more experience is looking at this, I will see what I can do on the general task of updating the code base including  libraries and deprecated Java features.

If anyone was already tackling any parts of this (or would like to), let me know so we can coordinate efforts.

maehne, 2020-01-29:
@davidhutchens: Thanks for offering your help in this area! It is well appreciated! Fixing the deprecation warnings and removing dependency on these old JARs will help to improve code quality. Currently, @BFH-ktt1 and myself have little bandwidth to spend on Logisim-evolution due to the exam period and preparation of a new course for the next semester.

One additional thing to investigate in the area of modernising the source code and on the Gradle side is, whether it is possible to modularise with reasonable effort the source code to enable usage of jlink directly package Logisim-evolution with a minimal JRE. This packaging should be handled by Gradle. It would be also nice to improve packaging for Windows and Linux using the new jpackage tool, which will be part of JDK14. Then, we could finally offer again an MSI installer package for Windows and DEB/RPM packages for Linux to facilitate installation of Logisim-evolution. We lost this ability when jarpackager got removed without replacement in JDK9 and we switched from Ant to Gradle. I recently found the [Badass JLink Plugin](https://github.com/beryx/badass-jlink-plugin) and the [Badass Runtime Plugin](https://badass-runtime-plugin.beryx.org/releases/latest/), which might help in this area. If you would like to give it a try, let me know. Within my limited bandwidth, I will be happy to help where I can, e.g., with testing.

davidhutchens, 2021-06-24:
With #719 this task is complete, other than #1 in the comment by @BFH-ktt1. How should we acknowledge the maven libraries that we use? We could add them to the list of contributors in the README. We could also add them to the scroll in the About panel. Thoughts?

maehne, 2021-07-01:
I have created issue #749 to track the remaining point. Let's therefore close this issue and continue the discussion there.

closed 2021-07-01 by maehne