logisim-evolution/logisim-evolution #324: Diode

edsonacordi, 2020-01-17:
It would be nice if it were possible to use diodes to implement logical ports.

maehne, 2020-01-17:
Do I understand you correctly that you would like to illustrate the functioning of simple diode logic gates as, e.g., described in [this Wikipedia article](https://en.wikipedia.org/wiki/Diode_logic)? You can fake already this behaviour using a PMOS with its gate connected to GND for the diode connecting the input to the pull-down resistor and an NMOS with its gate connected to VDD for the diode connecting the input to the pull-up resistor. You can hide this by creating a subcircuit for both diode connection cases: 

[image]

The ZIP-archive attached to this message contains a circuit file illustrating this approach. However, be aware that this is a quite crude model. I would rather illustrate the functioning of the different logic families using schematics for a SPICE simulator. E.g., LTspice is a suitable tool for this.

Based on the Java code already implementing the MOSFET primitive in Logisim, it should be relatively easy to implement this kind of diode model in Logisim. A PR is welcome.

[diode_logic.zip](https://github.com/reds-heig/logisim-evolution/files/4077334/diode_logic.zip)

edsonacordi, 2020-01-18:
Hello maehne, thank you for your reply.
Indeed, using a subcircuit to hide the transistor, your solution is satisfactory. I use LTspice and Qucs for simulations involving analog electronics. Thanks for sharing!

FamiliarBreakfast, 2020-02-13:
Surely a [buffer](https://i.imgur.com/ZS7YbLy.png) would do the trick? As if I understand it correctly, you're using it for one way flow, and it's much smaller than the above solution.

maehne, 2020-02-14:
Feel free to provide a circuit demonstrating how you would use a buffer to model the above diode logic gates. The tricky thing is that the output of the diode needs to be either '1' or 'Z' for the OR case or '0' or 'Z' for the AND case. I don't see a way to achieve this behaviour with a buffer.

closed 2021-07-13 by maehne