logisim-evolution/logisim-evolution #339: OR gate and button into AND gate not working correctly

Timshoo, 2020-02-06:
I have 2 comparators connected to an OR gate that feeds an AND gate with a button. The output of the AND gate clocks data int a register. My problem is that data clocks into the register regardless of the states of the OR or the AND gate whenever the button is pressed.
  It's not a very big circuit, just running a counter. When the button is pressed, the counter value shifts into Areg while the value of Areg is shifted to Breg(at the same time it increments a RAM address counter and shifts the main counter value into the RAM).  Both are fed into a subtractor. The comparators are used to store the SMALLEST COUNT. The first comp outputs true if the RAM address is less than 2(the very first button press will be 0 so wait for the second one), the second comp outputs true if the subtraction is less than the SMALLEST COUNT. 
  With both comps outputting false the OR gate is low so inA of the AND gate is low, but the AND gate is still letting a pulse through.
[image]

BFH-ktt1, 2020-02-06:
Dear @Timshoo,
The problem here is not related to logisim, but to your design strategy. You use a method which is know as gated clock, hence combinational logic that is creating a clock signal. This would work if the gates were ideal; however, gates have gate delay's (also simplified modeled in logisim), that can cause hazards (hence changes from 0->1 or 1->0 that in theory do not exist) causing the register to see positive edges where you would not expect them. The solution to this problem is to clock all synchronous elements with a clock and make a logic that synchronously clocks in the values.
I'll close this issue, please don't hesitate to ask further questions.

closed 2020-02-06 by BFH-ktt1

Timshoo, 2020-02-06:
A lot of stuff just clicked with that explanation. I replaced the wire to the clock input on the Shortest_interval register with a direct clock signal.

Thanks