logisim-evolution/logisim-evolution #340: A workaround for bidirectional/inout pins

eddiewastaken, 2020-02-10:
Here's an implemention I made of the [74LS245 (Bus Transceivers)](https://www.ti.com/lit/ds/symlink/sn54ls245-sp.pdf). The direction control makes it easy to reverse the direction without conflicting signals. Obviously in AND out pins had to be used, but this is the neatest solution I could manage for now:

[image]

Hopefully this helps someone!

ReGuess, 2020-05-19:
A slight improvement, making use of components' variable bit-width:

[image]
[bidirectional.zip](https://github.com/reds-heig/logisim-evolution/files/4647935/bidirectional.zip)

TChapman500, 2021-05-23:
You only need one group of tri-state buffers.

maehne, 2021-07-12:
Support for bidirectional pins is tracked in issue #253. I convert this issue describing possible workarounds to a discussion.

closed 2021-07-12 by maehne