logisim-evolution/logisim-evolution #342: Bug V3.3 option -tty

mbaillif, 2020-02-12:
Version 3.3 has a bug in the command line execution with the -tty option.
-tty -load m1.hex
-tty -table

No matter the circuit

[image]

klenin, 2020-02-22:
I have just downloaded logsim-evolution as a potential replacement for logisim and encountered this bug.
Brief analysis suggests that it is due to the fact that `WIDTH_ERROR_COLOR` is not initialized in tty mode, which causes `darker()` call to fail.
Possible fixes may be either to move global colors' initialization form `Value.getColor` to static initializer, or, conversely, replace static initializer of `Pin.ICON_WIDTH_COLOR` with dynamic one.

maehne, 2021-07-13:
This seems to have been fixed by PR #351. @mbaillif: Feel free to reopen if it should not be the case.

closed 2021-07-13 by maehne